introlab/rtabmap #653: Symbol Lookup Error: rtabmapviz

s3668134, 2020-12-03:
Hello,

After executing the steps in this post [https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817](url) to get rtabmapviz to work for rtabmap_ros. I was given the following error `/rtabmap_ros/rtabmapviz: symbol lookup error: /usr/local/lib/librtabmapcore.so.0.20: undefined symbol _ZNK3pcl6search6SearchINS_8PointXYZEE7getNameEv` 

Currently, the pcl version that I was using is PCL 1.8.0 and I know that this symbol is mentioned in libpcl_segmentation.so.1.8.1 but I am unsure as to how I could resolve this issue. If there is any tips that could be given, that would be great.

Thank you in advance.

hariharan382, 2022-08-13:
have you solved it? please post the solution

matlabbe, 2022-11-06:
I strongly recommend to use rviz when developing with jetson. I cannot keep up with latest jetpack releases (too much time consuming to rebuild/test everything everytime). Note that rtabmapviz should work on a remote computer, this is generally my way to visualize rtabmap when using jetson onboard a robot (and at the same time you don't waste precious CPU/GPU usage on the jetson just for visualization).