introlab/rtabmap #654: segmentation fault when trying to run c++ rgbd-mapping on macos

John-Trager, 2020-12-04:
There seems to be an issue when trying to run the c++ rgbd-mapping example. I get the following terminal ouput:
<code>
John@lab-1 tests2 % ./noEventsExample 20 2 10 stereo_20Hz stereo_20Hz stereo_20Hz/left stereo_20Hz/right 
objc[7770]: Class vtkCocoaTimer is implemented in both /usr/local/opt/vtk/lib/libvtkRenderingUI-9.0.1.dylib (0x12bc61438) and /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122fe90). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaTimer is implemented in both /usr/local/opt/vtk/lib/libvtkRenderingUI-9.0.1.dylib (0x12bc61438) and /opt/local/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x1114eee90). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaFullScreenWindow is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff08) and /opt/local/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x1114eef08). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaServer is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff30) and /opt/local/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x1114eef30). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaGLView is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff80) and /opt/local/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x1114eef80). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaFullScreenWindow is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff08) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x12bbaf240). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaServer is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff30) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x12bbaf268). One of the two will be used. Which one is undefined.
objc[7770]: Class vtkCocoaGLView is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x12122ff80) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x12bbaf2b8). One of the two will be used. Which one is undefined.
objc[7770]: Class QMacAutoReleasePoolTracker is implemented in both /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (0x120918ab8) and /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (0x1121333f0). One of the two will be used. Which one is undefined.
objc[7770]: Class QT_ROOT_LEVEL_POOL__THESE_OBJECTS_WILL_BE_RELEASED_WHEN_QAPP_GOES_OUT_OF_SCOPE is implemented in both /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (0x120918b30) and /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (0x112133468). One of the two will be used. Which one is undefined.
objc[7770]: Class KeyValueObserver is implemented in both /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (0x120918b58) and /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (0x112133490). One of the two will be used. Which one is undefined.
objc[7770]: Class RunLoopModeTracker is implemented in both /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (0x120918ba8) and /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (0x1121334e0). One of the two will be used. Which one is undefined.
zsh: segmentation fault
</code>

I think this has something to do with how rtab was built however I am unsure.
Thanks for the help.

matlabbe, 2020-12-10:
Did you use macport? https://github.com/introlab/rtabmap/wiki/Installation#source

John-Trager, 2020-12-11:
I used brew. Could that be an issue?

rtwomey, 2020-12-20:
@John-Trager I found a similar issue with brew. I believe vtk@8.2 is a dependency from [brew's pcl](https://formulae.brew.sh/formula/pcl), but the latest [OpenCV has a dependency on vtk@9](https://formulae.brew.sh/formula/opencv#default). With both installed, I suspect this is causing the segfault.

matlabbe, 2020-12-24:
The [opencv@3](https://formulae.brew.sh/formula/opencv@3#default) is independent of VTK, maybe it could help if it is the problem. I think with:
```bash
brew install sqlite pcl opencv@3
```
you would have the minimal dependencies to build rtabmap. I recommend also to build GTSAM or g2o for better graph optimization.

John-Trager, 2021-01-05:
@matlabbe okay, I will try this later this week thanks!