introlab/rtabmap #657: ZED 2 camera not recognised on v0.20.3

ManChrys, 2020-12-09:
Hello,

i tried to use new version (v0.20.3)  with my ZED2 camera but the zed sdk option in camera detection is gray. I tried the previous version (v0.20.2) and worked fine. 
I'm using ZED SDK 3.3.2 cuda 11 and windows 10.

Thanks in advance.

ManChrys, 2020-12-09:
hmmm ok i think i understand whats going on here. Is the problem cuda version ?

It works on windows 10 with ZED_SDK_v3.3.3_cuda10.2 but it doesnt work with ZED SDK 3.3.2 cuda 11. So the only difference here is cuda version. Correct me if im wrong.

Thanks in advance.

Myzhar, 2020-12-09:
@ManChrys when you download the ZED SDK you must choose the file that matches the version of CUDA installed on your machine.
If ZED_SDK_v3.3.3_cuda10.2 works that means that you installed CUDA 10.2, not CUDA11

ManChrys, 2020-12-09:
Hi @Myzhar,
first of all you did great job guys here!!
I have one pc with windows 10 and cuda version 10.2 and RTABMAP works fine.
On my another pc i have windows 10 and cuda version 11.0 and RTABMAP doesnt work.
So the question is RTABMAP is compatible with  cuda 11 or i have to work only on cuda 10.2?

Thanks in advance.

matlabbe, 2020-12-10:
We have to rebuild the binaries for cuda 11. I'll try to do this soon.

ManChrys, 2020-12-10:
@matlabbe thank you guys. I love your project :D

ManChrys, 2020-12-14:
Hello guys,

i've installed rtabmap 0.20.7 on windows 10 with cuda 11.0,
i've installed  ZED_SDK_Windows10_cuda11.0_v3.3.2

but zed sdk option still not available.

Zed tools (Zedfu, ..)  are working properly so my ZED2 is recognized.

Thanks in advance.

matlabbe, 2020-12-14:
I update the release notes, RTAB-Map has been built with CUDA 11.1 and ZED 3.3.3. Normally even if you didn't install those, the menu option should be enabled:
[image]

Make sure in About dialog it says "Yes" for "With stereo Zed":
[image]

I tested with Zed1 and Zed-mini on a different computer than my build machine. I don't have a Zed2 so I cannot say if there is a difference, but the menu option should still be enabled. If there is an incompatibility with zed2, it could be more an error when starting the camera.

ManChrys, 2020-12-16:
Thnx for the reply. 
In About dialog it says with stereo Zed : No

how can i change this ?

matlabbe, 2020-12-16:
You may not be starting the right rtabmap, what is the version shown? is it 0.20.7? Make sure you are using the cuda11 build, which is the only one with zed sdk support.

ManChrys, 2020-12-16:
Hi
First of all thank you guys for the quick responses!! 
The problem was that after your update i accidentally dowloaded RTABMap-0.20.7-win64.exe instead of RTABMap-0.20.7-win64-cuda11_1.exe.

Now it worked fine!! ty guys!!

[image]
[image]

closed 2020-12-16 by ManChrys