introlab/rtabmap #661: dataRecorder tool configuration and L515

rlabs-oss, 2020-12-13:
On starting the dataRecorder, RS2_OPTION_GLOBAL_TIME_ENABLED is set back to 1 (was 0) 

This causes the RealSense L515 to stall with the missing frames error as previously worked around in the GUI option. 

For low end systems using the recorder to capture and post-process, it would be desirable to support
an option to load the config.ini or use the RealSense2\global_time_sync value.

matlabbe, 2020-12-13:
I modified the interface of rtabmap-databaseRecorder to use a config file to set parameters. The config file is the same format than the one of the main interface, so open RTAB-Map->Preferences->Source, set camera parameters then save the settings (Preferences->General Settings->Save Settings (*.ini)). That way, all parameters in Source can be used with the command line tool.

rlabs-oss, 2020-12-13:
Excellent update - Big Thank you.

closed 2020-12-13 by rlabs-oss