introlab/rtabmap #662: Mac binary hanging on Big Sur

rtwomey, 2020-12-20:
I found that running the latest MacOS binary release v0.20.7 on Big Sur (Version 11.1) causes a hang where the CPU usage is at 100%. Running in the terminal shows:

```
➜  /Applications ./RTABMap.app/Contents/MacOS/RTABMap --udebug
[DEBUG] (2020-12-19 22:44:57.075) MainWindow.cpp:182::MainWindow()
```

I believe the issue is the Qt version it's linked against (otool reports 5.14.2), which may have a [bug with Bug Sur](https://bugreports.qt.io/browse/QTBUG-87014). The following workaround, [inspired by this post](https://stackoverflow.com/a/64878899), has worked for me:

```
➜  /Applications QT_MAC_WANTS_LAYER=1 ./RTABMap.app/Contents/MacOS/RTABMap --udebug
[DEBUG] (2020-12-19 22:51:09.559) MainWindow.cpp:182::MainWindow()
[DEBUG] (2020-12-19 22:51:09.804) MainWindow.cpp:209::MainWindow() Setup ui...
...
```

I think the solution will be to build against Qt version 5.15 or later to resolve the issue, but I have not built it myself yet.

matlabbe, 2020-12-20:
Thx for the info, I thought there was an incompatibility on my mac mini M1 for some reasons, but with QT_MAC_WANTS_LAYER=1, it works now on M1 (rosetta2) too. I'll try to update Qt on my macbook pro 2010 and regenerate the DMG.

matlabbe, 2020-12-21:
I updated qt5 with macport with latest qt 5.15.2, the bug is still there. I think the bug has been patched the same day, just after 5.15.2 release:
19 nov 2020 13:16: https://download.qt.io/official_releases/qt/5.15/5.15.2/single/
19 nov 2020 14:21: https://bugreports.qt.io/browse/QTBUG-87014

I am not sure I want to rebuild everything manually from source, hopefully 5.15.3 may be released soon. I added the command line with the workaround on the [Release](https://github.com/introlab/rtabmap/releases/tag/0.20.7) page.

closed 2020-12-21 by matlabbe