introlab/rtabmap #663: Import sensor data, sent from rpi

spacemonk-ey, 2020-12-21:
Hi,
I was wondering if it's possible to use rtabmap on a windows pc and use sensor data, that is sent via wifi from a rpi4 to which a d415 and a t265 are connected.
I actually just started to work with rpi and rgbd cameras as part of my thesis and I'm searching for way to visualize a pointcloud map on a external pc, while a drone with the rpi and the cameras onboard is moving around.
Any suggestions?
Thanks in advance.

matlabbe, 2020-12-24:
I suggest to install a virtual machine with Ubuntu/ROS or dualboot with Ubuntu/ROS. After that, it is mostly subscribing right topics published from the Pi on the PC: http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping

spacemonk-ey, 2021-01-09:
Thanks, I'm gonna try that.