introlab/rtabmap #664: Error while running make command

AyanHBK123, 2020-12-25:
When running make command this error is encountered:

/usr/bin/ld: ../../../bin/librtabmap_core.so.0.19.3: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
/usr/bin/ld: ../../../bin/librtabmap_gui.so.0.19.3: undefined reference to `pcl::search::Search<pcl::PointXYZRGBNormal>::getName() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.19.3: undefined reference to `pcl::search::Search<pcl::PointXYZI>::getName() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.19.3: undefined reference to `pcl::search::Search<pcl::PointXYZRGB>::getName() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.19.3: undefined reference to `pcl::search::Search<pcl::PointNormal>::getName() const'
collect2: error: ld returned 1 exit status
make[2]: *** [app/src/CMakeFiles/rtabmap.dir/build.make:385: ../bin/rtabmap] Error 1
make[1]: *** [CMakeFiles/Makefile2:422: app/src/CMakeFiles/rtabmap.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

matlabbe, 2020-12-25:
0.19.3, what is your setup? it would be better to use latest version of rtabmap.

AyanHBK123, 2020-12-25:
I am installing on Pi 4 on Raspbian Buster. 
Is there a official guide for this ??

matlabbe, 2020-12-25:
I did install rtabmap on Ubuntu Mate 20.04, but not on raspbian. There is no official guide, normally it shoudl be simialr to any linux distributions. The problem could come from which library versions are installed. First, 0.19.3 is quite old. I would update to latest version to see if that error will go away.