introlab/rtabmap #665: Userdatacompressed is over written in Memory::createSignature?

RigerLee, 2020-12-26:
Hi, I tried to send OdometryEvent with SensorData where I have set userdatacompressed (CV_8UC1, row=1). Then I found it missing after I receive RtabmapEvent.

I noticed that after [this line](https://github.com/introlab/rtabmap/blob/master/corelib/src/Memory.cpp#L5021), if data.userDataRaw().empty() == true, userdatacompressed will be overwritten by compressedUserData. Is it deliberately designed like this?

matlabbe, 2020-12-27:
See
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/src/SensorData.cpp#L386-L389
and
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/include/rtabmap/core/SensorData.h#L218-L228

Can you be more specific of which line is causing the problem? Only if CV_8UC1, row=1, col>=3 is considered as compressed. How did you compress the data? If you pass compressed data, the use data (raw and compressed) will be cleared, keeping only the compressed one (the raw data will be gone).

RigerLee, 2020-12-28:
Well, let me be more specific.

Suppose I use compressed jpg(CV_8UC1, row=1, col>3) as user data to construct the SensorData(for some reason, the SensorData.imageRaw() will store another image), then SensorData.userDataRaw().empty() == true and SensorData.userDataCompressed().empty() == false.

Then in Memory::createSignature, when constructing Signature from SensorData, suppose this->isBinDataKept() == true.
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/src/Memory.cpp#L5039-L5080
The above code will compress data.userDataRaw() anyway, which is empty, then compressedUserData becomes empty.
here:
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/src/Memory.cpp#L5073
and here:
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/src/Memory.cpp#L5080

And then the original userdata is assigned back
https://github.com/introlab/rtabmap/blob/da99d7e4a0a24edc5552ad31e2cc829f0ebe692d/corelib/src/Memory.cpp#L5231

After the above steps, SensorData with only compressed userdata will loss its compressed userdata. So I think maybe you could simply add one more check here to handle this case. Will there be any side effects?