introlab/rtabmap #666: Algrithm of Dense reconstruction in RTABMAP？ t

liuzhenboo, 2020-12-27:
I have practiced your project  for one year, it's a very large and excellent project！
Now I wonder the algrithm of dense reconstruction in RTABMAP？ 

[image]

I have read your paper of this project, but I didn't find some statement of the dense reconstruction algorithm? So I open the issue to asking for your help. 
Can you introduce your algorithm？Can you provide me some documents about your dense reconstruction algorithm? Thanks a lot sincerely.

matlabbe, 2020-12-27:
By dense reconstruction, do you mean dense point cloud? If so, it just takes the depth image (or disparity image in case of stereo) and projects it in 3D. In the case of stereo, opencv's StereoBM is used by default.

liuzhenboo, 2020-12-28:
> By dense reconstruction, do you mean dense point cloud? If so, it just takes the depth image (or disparity image in case of stereo) and projects it in 3D. In the case of stereo, opencv's StereoBM is used by default.

I get it, thank you very much!