introlab/rtabmap #667: T265 Calibration issues

VisibleMeasurement, 2020-12-27:
I have been trying to calibrate the T265 on Windows 10 with the 20.7 standalone.  I have a few issues:

I cannot get the "Calibrate" button to become pressable without first checking the Unlock option.  I had up to 150 images and it still wasn't working.

Then if I hit calibrate it freezes without saving.  I even left it "working" over night.  

This is eating up a bunch of time, any ideas?

matlabbe, 2020-12-27:
Fisheye stereo cameras are more difficult to calibrate. For T265, we have to "unlock" as the size cannot be reached (because it is fisheye). In this post I have 60 templates:
http://official-rtab-map-forum.67519.x6.nabble.com/Slam-using-Intel-RealSense-tracking-camera-T265-tp6333p6336.html

I don't recommend to go over 60-80 templates, as calibration becomes exponentially long to compute and may fail. Here is he patterns I do:
- Zoom in and out
- move up and down with same angle (close)
- move up and down with same angle (far)
- move right and left with same angle (close)
- move right and left with same angle (far)
- do the skew
- unlock and calibrate (even if you have 30 templates)

VisibleMeasurement, 2020-12-27:
OK great.  Thanks.  I got it to save.  I think there was an issues where an earlier (and poor)  attempt to calibration saved with the serial number of the device.  When I went back to export the calibration, it froze and prevented later calibrations from saving.

Thanks for the help!

closed 2020-12-27 by VisibleMeasurement