introlab/rtabmap #668: Synchronizing T265 with L515

VisibleMeasurement, 2020-12-27:
I am on Windows 10 with Standalone version 20.7 and current RealSense firmware on 2.40.0.

After calibrating the T265, I am trying to run the L515 with the T265 I get the following error:

`[ WARN] (2020-12-27 12:49:16) CameraRealSense2.cpp:288::rtabmap::CameraRealSense2::getPoseAndIMU() Could not find poses to interpolate at image time 262985.759000 (earliest is 1609091352154.697998). Are sensors synchronized?`

How can I sync them?  Is the fact in Device Manger the T265 keeps changing back and forth to "Movidius MA2x5x" offer a clue?  I'm working on a Surface Pro 7 and having issues with the USB ports not providing sufficient power and occasionally disconnecting.  I have some different cables on order that may help.

matlabbe, 2020-12-27:
In the log, we can see that both sensors use different time stamps: 262985.759000 versus 1609091352154.697998. You may disable Global Time Sync option in Source->RGBD->RealSense2 section. Note that dual mode between T265 and L515 has never been tested (I don't have a L515...).

VisibleMeasurement, 2020-12-27:
Thanks so much for the help.  Unfortunately, the same error comes up.  And, of course, the T265 doesn't do much on its own.  I read a few posts from a user that had them working together, but they seemed to be going through ROS with more settings modifications.  I'll play around with just the L515 for now and see how its results compare with the L515 in Dot3D pro that I'd previously tested.  Thanks for all your hard work on this, it's an amazing tool to have.

closed 2020-12-27 by VisibleMeasurement

matlabbe, 2021-02-13:
Just stumbled again on this issue, the T265 + L515 integration works on latest builds ( standalone version). In this [release](https://github.com/introlab/rtabmap/releases/tag/0.20.8), you can see some info on how using L515 in ICP mode, which can be good in environment with a lot of geometry or in dark rooms.