introlab/rtabmap #669: A version of the docker image with libfreenect2 installed

jfm535, 2020-12-30:
I created a version of the focal docker file which should support the kinectv2 using freeenect2.
[Dockerfile.txt](https://github.com/introlab/rtabmap/files/5755207/Dockerfile.txt)

I combined the docker information from here https://github.com/nucleardreamer/docker-libfreenect2 

and adjusted it to keep the number of layers low.

matlabbe, 2020-12-30:
Hi,

I think you could have been starting `FROM introlab3it/rtabmap:focal` (which depends also from `ros/noetic-perception`), to avoid rebuilding gtsam and libpointmatcher:
```dockerfile
FROM introlab3it/rtabmap:focal

# Install build dependencies
RUN apt-get update && \
    apt-get install -y \
    beignet-dev \
    build-essential \
    cmake \
    git \
    libusb-1.0.0-dev \
    libturbojpeg \
    libturbojpeg0-dev \
    libglfw3-dev \
    libopenni2-dev \
    opencl-headers \
    openni2-utils \
    pkg-config \
    udev && \
    rm -rf /var/lib/apt/lists/

WORKDIR /root/

# Clone source code
ARG CACHE_DATE=2016-01-01
RUN git clone https://github.com/introlab/rtabmap.git

RUN rm /bin/sh && ln -s /bin/bash /bin/sh

# Build RTAB-Map project
RUN source /ros_entrypoint.sh && \
    cd rtabmap/build && \
    cmake .. -Dfreenect2_DIR=/root/freenect2/lib/cmake/freenect2 && \
    make && \
    make install && \
    cd ../.. && \
    rm -rf rtabmap && \
    ldconfig

```

jfm535, 2020-12-30:
Going to try your file I kept having segmentation faults

jfm535, 2020-12-31:
yes your docker file worked with just some minor changes to support intel

```
FROM introlab3it/rtabmap:focal

# Install build dependencies
RUN apt-get update && \
    apt-get install -y \
    beignet-dev \
    build-essential \
    cmake \
    git \
    libusb-1.0.0-dev \
    libturbojpeg \
    libturbojpeg0-dev \
    libglfw3-dev \
    libopenni2-dev \
    opencl-headers \
    openni2-utils \
    pkg-config \
    libgl1-mesa-glx \
    libgl1-mesa-dri \
    mesa-utils \
    udev && \
    rm -rf /var/lib/apt/lists/

WORKDIR /root/

RUN git clone https://github.com/OpenKinect/libfreenect2.git \
    && cd libfreenect2 \
    && mkdir build && cd build \
    && cmake .. -DBUILD_OPENNI2_DRIVER=ON -DCMAKE_INSTALL_PREFIX=/opt/ros/noetic \
    && make \
    && make install \
    && cp /root/libfreenect2/platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/ \
    && ldconfig /root/freenect2 \
    && ln -s /libfreenect2/build/bin/Protonect /usr/local/bin/kinect_test \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

# Clone source code
ARG CACHE_DATE=2016-01-01
RUN git clone https://github.com/introlab/rtabmap.git

RUN rm /bin/sh && ln -s /bin/bash /bin/sh

# Build RTAB-Map project
RUN source /ros_entrypoint.sh && \
    cd rtabmap/build && \
    cmake .. && \
    make && \
    make install && \
    cd ../.. && \
    rm -rf rtabmap && \
    ldconfig
```

And heres the script i used to be able to get it to run

```
#!/bin/bash
xhost +
docker run -it \
--volume=/tmp/.X11-unix:/tmp/.X11-unix \
--device=/dev/dri:/dev/dri \
--env="DISPLAY=$DISPLAY" \
--network=host \
--device /dev/bus/usb:/dev/bus/usb \
rtabmap-freenect2
```
for whatever reason the most important part seems to be the `--network=host` without it the docker image cant access the xserver.