introlab/rtabmap #670: The map constructed by rtabmap is tilted

kotemen-master, 2021-01-04:
Hello

I would like to use rtabmap in a gazebo simulation.

The map constructed by rtabmap is tilted in the pitch direction. 
[image]

When the base_link with realsense(camera_link) is tilted in the pitch direction, the map is also rotated in the pitch direction.
[image]

A similar tilted map can be seen in rtabmapviz.
Could you give me some advice on the cause?

My environment is as follows.
ros melodic
gazebo version9 

I have a realsense D435 attached to the front of the robot with a 45 degree pitch tilt. 
I am using the realsense2_description urdf to launch the realsense D435 on gazebo. I am getting the CAD data of the realsense D435 and the gazebo settings from realsense2_description.
https://github.com/pal-robotics-forks/realsense/tree/upstream/realsense2_description/urdf

I am using realsense_gazebo_plugin as a gazebo plugin to use realsense D435 on gazebo.
https://github.com/pal-robotics/realsense_gazebo_plugin.git

> When the whole map is tilted, it means the camera is tilted in reality but not in TF (or vice-versa).

↑The above comment is the admin's response to a similar question by another questioner.
http://official-rtab-map-forum.67519.x6.nabble.com/Illumination-issues-with-RTAB-map-td5335.html
However, I can confirm from rviz that the camera is tilted. base_link and camera_link have a pitch angle of 45 degrees tilted.
In addition, when the actual robot was operated using the same urdf file and lanucn file, there was no tilt in the map. The only major difference between the actual robot and the gazebo simulation is that there is no world frame in the case of the real machine experiment.

The virtual realsense on gazebo also seems to be working correctly because it is getting the correct point cloud.
In the above figures, the gray point clouds were generated by virtual realsense on gazebo.

I am attaching the TF tree.
[TFtree_gazebo_simulation.pdf](https://github.com/introlab/rtabmap/files/5765359/TFtree_gazebo_simulation.pdf)

Also, I am attaching my launch file. I have converted the .launch file to a .txt file.
[rtabmap_sim.txt](https://github.com/introlab/rtabmap/files/5765816/rtabmap_sim.txt)

Please let me know if there is any information I am missing.

Thank you.

matlabbe, 2021-01-11:
You said "base_link and camera_link have a pitch angle of 45 degrees tilted.", and your launch is using base_link as frame_id. If base_link is tilted, the resulting map will be tilted. There should be a 45 deg difference between base_link and the camera link. Base_link should not have any rotation.

kotemen-master, 2021-01-15:
@matlabbe 

Thank you for answering my question. Thanks to your answer, I understand what I need to think about to solve this problem.
The cause was that the depth image was not registered to RGB image. Unlike rs_align_depth.launch (for actual realsense), the unregistered depth image(/camera/depth/color/image_raw) was being provided to rtabmap and rgbd_odometry.

Especially, another answer of yours helped me a lot.
http://official-rtab-map-forum.67519.x6.nabble.com/3D-mappping-using-D435-not-able-to-get-a-proper-map-td7001.html

Thank you very much.

closed 2021-01-15 by kotemen-master