introlab/rtabmap #671: Saving post-processed graph to the database

aashutosh1997, 2021-01-08:
I have been trying to save the post processed graph to a database using desktop gui, but couldn't find a databaseViewer or any option in the 'Edit' Menu, like mentioned here (http://official-rtab-map-forum.67519.x6.nabble.com/Save-post-processed-database-td2334.html). Cannot figure out what I am missing here, thanks for the help in advance!

sumitsarkar1, 2021-01-27:
If you have installed rtabmap correctly then just typing rtabmap-databaseViewer in the terminal will open the application just like rtabmap opens. Or you will find it in the bin folder `rtabmap/bin` . However I did not I find any **post-processing** tab under the edit menu of rtabmap-databaseViewer as it is under the **tools** tab of rtabmap . So like..how to save the post processing changes to the .db file so that the post processed file gets loaded next time we open the .db file.

matlabbe, 2021-02-12:
[image]

The post-processing options are in Edit menu of rtabmap-databaseViewer. You can also open DatabaseViewer from Tools->Edit database... of RTAB-Map standalone app.

To save the changes, you have to close the database, then if you made changes, you will be asked to save them or not.

aashutosh1997, 2021-02-22:
Thank you for the help, something to be noted here is that Tools->Edit Database... option is disabled if a database is already open. 

Also, I could not find "Refine Links with ICP" and "Sparse Bundle Adjustment (SBA)" options in the databaseViewer, any note on that @matlabbe? 

PS: Must be frustrating for you to answer these dumb questions, I really appreciate the help.

matlabbe, 2021-02-26:
`Refine Links with ICP` can be done with Edit -> Refine... options. To refine with ICP, make sure to open Core Parameters panel and set Reg/Strategy to 1.

The global SBA optimization cannot be done in databaseViewer.

OlegAlexander, 2021-09-08:
Dear @matlabbe, 

Thank you for creating RTAB-Map.

I'm also struggling with saving the sparse bundle adjustment (SBA) optimization to the database from RTAB-Map standalone or from databaseViewer. As you said, SBA isn't available in the databaseViewer. But I can't figure out how to save it from RTAB-Map standalone either. 

The directions in the standalone Post-Processing dialog say: _"Post-processing only affects visualization in the GUI and the export actions. To actually push those optimizations to the database, use Database Viewer tool (Tools->Edit database)"_. So I tried the following workflow:

1. Open the database.
2. Run SBA. 
3. But Tools->Edit database is grayed out when a database is loaded.

So I tried:

1. Open the database.
2. Close the database, which will make the Tools->Edit database button available, but keep the database in RAM.
3. Run SBA.
4. Now I go to Tools->Edit database and it brings up a file selection dialog.
5. I choose the database file and it opens in the databaseViewer but without the SBA changes.

Finally, I tried:

1. Open the database.
2. Run SBA.
3. Close the database, but the SBA changes are not saved.

I've also looked at all the command line tools in RTABMap/bin. The only one that mentions bundle adjustment is `rtabmap-export.exe`. I tried it and it looked like it was going to work, but then it failed at the end:

```console
C:\Program Files\RTABMap\bin>rtabmap-export.exe --ba --save_in_db "C:\Users\oalexand\Documents\RTAB-Map\secondFloor_01 - Copy.db"
Loading database "C:\Users\oalexand\Documents\RTAB-Map\secondFloor_01 - Copy.db"...
Loading database "C:\Users\oalexand\Documents\RTAB-Map\secondFloor_01 - Copy.db"... done (9.620410s).
Optimizing the map...
Optimizing the map... done (1.017754s, poses=305).
Global bundle adjustment...
[FATAL] (2021-09-07 16:24:34) main.cpp:402::main() Condition (node.sensorData().cameraModels().size()==1) not met!
```

So my question is: how can I save the SBA optimization to the database? Thank you!

matlabbe, 2021-09-09:
Global SBA result cannot be saved back to database, only in exported data. Do you have more than one camera? SBA multi-camera support has not been implemented.

OlegAlexander, 2021-09-09:
Thank you for your quick response. No, only a single camera. The RealSense T265. 

Is saving the SBA result to the database a planned feature? 

Could you please explain the main difference between "Detect more loop closures" and SBA? Thank you.

matlabbe, 2021-09-09:
Hi,

The rtabmap-export tool didn't support stereo data, I updated it in this [commit](https://github.com/introlab/rtabmap/commit/263e0170f154aae5b5442fd303229aec1a488200).

Saving back the SBA optimization was not planned, as rtabmap re-optimizes the graph when querying the map. Currently, even if we save back the optimized poses, they would be used only in localization mode. 

"Detect more loop closures" detects more loop closures and add new constraints to graph between nodes (like loop closures detected online). SBA is optimizing the graph using also visual feature correspondences, instead of optimizing only the pose-graph.

matlabbe, 2021-09-11:
In this [commit](https://github.com/introlab/rtabmap_ros/commit/a324562ee07e528e5dfb87caa5d74903878471b4), I added some post-processing services to detect more loop closures or start a global bundle adjustment:
```bash
# To detect more loop closures:
rosservice call /rtabmap/detect_more_loop_closures "{cluster_radius_max: 1.0, cluster_radius_min: 0.0, cluster_angle: 30.0, iterations: 1, intra_only: false, inter_only: false}"

# To do global bundle adjustment:
# If you are in SLAM mode, make sure to pause rtabmap, 
# otherwise the changes made will be overwritten the next 
# time a new node is added to graph
rosservice call /rtabmap/pause
rosservice call /rtabmap/global_bundle_adjustment "{type: 0, iterations: 20, pixel_variance: 1.0, voc_matches: false}"
```

alexk1976, 2023-05-13:
Hi @matlabbe , whether GBA was implemented over multiple-cameras?