introlab/rtabmap #673: double free or corruption (out) on Docker focal image

parkerlreed, 2021-01-11:
Using the focal image from the repo with some customization (including libfreenect2) is causing it to segfault on running rtabmap

Arch Linux host
```
docker 1:20.10.2-1
```

Dockerfile
```
# Image: introlab3it/rtabmap:focal

FROM ros:noetic-perception

ENV DEBIAN_FRONTEND=noninteractive

# Install build dependencies
RUN apt-get update && \
    apt-get install -y git software-properties-common ros-noetic-rtabmap-ros && \
    apt-get remove -y ros-noetic-rtabmap && \
    rm -rf /var/lib/apt/lists/

WORKDIR /root/

# GTSAM
RUN add-apt-repository ppa:joseluisblancoc/gtsam-develop -y
RUN apt install libgtsam-dev

# libpointmatcher 
RUN git clone https://github.com/ethz-asl/libnabo.git
#commit Apr 25 2018
RUN cd libnabo && \
    git checkout 7e378f6765393462357b8b74d8dc8c5554542ae6 && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release .. && \
    make -j$(nproc) && \
    make install && \
    cd && \
    rm -r libnabo
RUN git clone https://github.com/ethz-asl/libpointmatcher.git
#commit Jan 19 2018
RUN cd libpointmatcher && \
    git checkout 00004bd41e44a1cf8de24ad87e4914760717cbcc && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release .. && \
    make -j$(nproc) && \
    make install && \
    cd && \
    rm -r libpointmatcher

RUN apt-get update
RUN apt-get install -y mesa-utils
RUN apt-get install -y xserver-xorg-video-all
RUN apt-get install -y libusb-1.0-0-dev
RUN apt-get install -y libturbojpeg0-dev
RUN apt-get install -y libglfw3-dev
RUN git clone https://github.com/OpenKinect/libfreenect2
RUN cd libfreenect2 && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make -j4 && \
    make install && \
    cd && \
    rm -r libfreenect2

# Clone source code
ARG CACHE_DATE=2016-01-01
RUN git clone https://github.com/introlab/rtabmap.git

RUN rm /bin/sh && ln -s /bin/bash /bin/sh

# Build RTAB-Map project
RUN source /ros_entrypoint.sh && \
    cd rtabmap/build && \
    cmake .. && \
    make && \
    make install && \
    cd ../.. && \
    rm -rf rtabmap && \
    ldconfig
```

```
[parker@wolfcola focal]$ xhost +
access control disabled, clients can connect from any host
[parker@wolfcola focal]$ docker run -it --rm    --privileged    --env="DISPLAY=$DISPLAY"    --env="QT_X11_NO_MITSHM=1"    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw"    --env="XAUTHORITY=$XAUTH"    --volume="$XAUTH:$XAUTH"    --volume="/dev/bus/usb:/dev/bus/usb"    rtabmap3d    rtabmap
double free or corruption (out)
```

```
[parker@wolfcola focal]$ coredumpctl info 23288
           PID: 23288 (rtabmap)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 6 (ABRT)
     Timestamp: Mon 2021-01-11 03:10:33 EST (3min 49s ago)
  Command Line: rtabmap
    Executable: /usr/local/bin/rtabmap
 Control Group: /system.slice/docker.service
          Unit: docker.service
         Slice: system.slice
       Boot ID: 01a64c5a05b9485a86c46dbc80f1cfe6
    Machine ID: bd9f838e9ab447dca03be703daacbd26
      Hostname: wolfcola
       Storage: /var/lib/systemd/coredump/core.rtabmap.0.01a64c5a05b9485a86c46dbc80f1cfe6.23288.1610352633000000.zst (inaccessible)
       Message: Process 23288 (rtabmap) of user 0 dumped core.
                
                Stack trace of thread 32:
                #0  0x00007f335147c18b n/a (/usr/lib/x86_64-linux-gnu/libc-2.31.so + 0x4618b)
                #1  0x00007f33514ce47c n/a (/usr/lib/x86_64-linux-gnu/libc-2.31.so + 0x9847c)
```

parkerlreed, 2021-01-11:
Currently testing with GTSAM disabled as mentioned here https://github.com/introlab/rtabmap/issues/505#issuecomment-589381659

Will update if this works.

parkerlreed, 2021-01-11:
Yep that did it 
[image]

Does the included gtsam need to be fixed? (if this even affects the published focal image and not just the locally built Dockerfile)

BorcherdVB-2055428, 2021-03-16:
I'm having the same problem. I've edited the Dockerfile, shown below, so it builds without gtsam, but i'm still getting the error "double free or corruption (out)". Any thoughts? Thanks!

```
# Image: introlab3it/rtabmap:focal

FROM ros:noetic-perception

ENV DEBIAN_FRONTEND=noninteractive

# nvidia-container-runtime
ENV NVIDIA_VISIBLE_DEVICES \
    ${NVIDIA_VISIBLE_DEVICES:-all}
ENV NVIDIA_DRIVER_CAPABILITIES \
    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics

# Install build dependencies
RUN apt-get update && \
    apt-get install -y git software-properties-common ros-noetic-rtabmap-ros && \
    apt-get remove -y ros-noetic-rtabmap && \
    rm -rf /var/lib/apt/lists/

WORKDIR /root/

# GTSAM
RUN add-apt-repository ppa:joseluisblancoc/gtsam-develop -y
RUN apt install libgtsam-dev

# libpointmatcher 
RUN git clone https://github.com/ethz-asl/libnabo.git
#commit Apr 25 2018
RUN cd libnabo && \
    git checkout 7e378f6765393462357b8b74d8dc8c5554542ae6 && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release .. && \
    make -j$(nproc) && \
    make install && \
    cd && \
    rm -r libnabo
RUN git clone https://github.com/ethz-asl/libpointmatcher.git
#commit Jan 19 2018
RUN cd libpointmatcher && \
    git checkout 00004bd41e44a1cf8de24ad87e4914760717cbcc && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_BUILD_TYPE=Release .. && \
    make -j$(nproc) && \
    make install && \
    cd && \
    rm -r libpointmatcher

RUN apt-get update
RUN apt-get install -y mesa-utils
RUN apt-get install -y xserver-xorg-video-all
RUN apt-get install -y libusb-1.0-0-dev
RUN apt-get install -y libturbojpeg0-dev
RUN apt-get install -y libglfw3-dev
RUN git clone https://github.com/OpenKinect/libfreenect2
RUN cd libfreenect2 && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make -j4 && \
    make install && \
    cd && \
    rm -r libfreenect2

# Clone source code
ARG CACHE_DATE=2016-01-01
RUN git clone https://github.com/introlab/rtabmap.git

RUN rm /bin/sh && ln -s /bin/bash /bin/sh

# Build RTAB-Map project
RUN source /ros_entrypoint.sh && \
    cd rtabmap/build && \
    cmake -DWITH_GTSAM=OFF .. && \
    make -j4 && \
    make install && \
    cd ../.. && \
    rm -rf rtabmap && \
    ldconfig
```

matlabbe, 2021-05-02:
The problem was the gtsam version. The 4.0 version from here https://gtsam.org/get_started/ is working. The [focal image](https://github.com/introlab/rtabmap/blob/master/docker/focal/Dockerfile) has been updated with the fix. I also added libfreenect2 and other camera sdks for convenience.

closed 2021-05-02 by matlabbe