introlab/rtabmap #674: OptimizerG2O compile time errors

parkerlreed, 2021-01-11:
Here's another stab at building on Arch Linux. 

g2o and gtsam built with march native disabled.

```
gcc 10.2.0-4
pcl 1.11.1-1
g2o-git r1271.bc440f4-1
gtsam 4.0.3-1
opencv 4.5.1-1
```

```
[parker@wolfcola build]$ make
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp: In member function ‘virtual std::map<int, rtabmap::Transform> rtabmap::OptimizerG2O::optimizeBA(int, const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap::Link>&, const std::map<int, rtabmap::CameraModel>&, std::map<int, cv::Point3_<float> >&, const std::map<int, std::map<int, rtabmap::FeatureBA> >&, std::set<int>*)’:
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1504:10: error: ‘VertexSBAPointXYZ’ is not a member of ‘g2o’; did you mean ‘VertexPointXYZ’?
 1504 |     g2o::VertexSBAPointXYZ* vpt3d = new g2o::VertexSBAPointXYZ();
      |          ^~~~~~~~~~~~~~~~~
      |          VertexPointXYZ
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1504:29: error: ‘vpt3d’ was not declared in this scope; did you mean ‘pt3d’?
 1504 |     g2o::VertexSBAPointXYZ* vpt3d = new g2o::VertexSBAPointXYZ();
      |                             ^~~~~
      |                             pt3d
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1504:41: error: expected type-specifier
 1504 |     g2o::VertexSBAPointXYZ* vpt3d = new g2o::VertexSBAPointXYZ();
      |                                         ^~~
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1682:14: error: ‘VertexSBAPointXYZ’ is not a member of ‘g2o’; did you mean ‘VertexPointXYZ’?
 1682 |       ((g2o::VertexSBAPointXYZ*)(*iter)->vertex(0))->setEstimate(Eigen::Vector3d(pt3d.x, pt3d.y, pt3d.z));
      |              ^~~~~~~~~~~~~~~~~
      |              VertexPointXYZ
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1682:32: error: expected primary-expression before ‘)’ token
 1682 |       ((g2o::VertexSBAPointXYZ*)(*iter)->vertex(0))->setEstimate(Eigen::Vector3d(pt3d.x, pt3d.y, pt3d.z));
      |                                ^
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1760:15: error: ‘VertexSBAPointXYZ’ in namespace ‘g2o’ does not name a type; did you mean ‘VertexPointXYZ’?
 1760 |    const g2o::VertexSBAPointXYZ* v;
      |               ^~~~~~~~~~~~~~~~~
      |               VertexPointXYZ
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1764:5: error: ‘v’ was not declared in this scope; did you mean ‘pcl::fields::v’?
 1764 |     v = (const g2o::VertexSBAPointXYZ*)optimizer.vertex(negVertexOffset + id*-1);
      |     ^
      |     pcl::fields::v
In file included from /usr/include/boost/preprocessor/variadic/elem.hpp:102,
                 from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:66,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.11/pcl/point_struct_traits.h:42,
                 from /usr/include/pcl-1.11/pcl/type_traits.h:40,
                 from /usr/include/pcl-1.11/pcl/memory.h:46,
                 from /usr/include/pcl-1.11/pcl/impl/point_types.hpp:41,
                 from /usr/include/pcl-1.11/pcl/point_types.h:349,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/Signature.h:32,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/Optimizer.h:37,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/optimizer/OptimizerG2O.h:33,
                 from /home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:37:
/usr/include/pcl-1.11/pcl/impl/point_types.hpp:1893:1: note: ‘pcl::fields::v’ declared here
 1893 | POINT_CLOUD_REGISTER_POINT_STRUCT (pcl::_PointXYZHSV,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1764:16: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
 1764 |     v = (const g2o::VertexSBAPointXYZ*)optimizer.vertex(negVertexOffset + id*-1);
      |                ^~~
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1764:10: error: expected primary-expression before ‘const’
 1764 |     v = (const g2o::VertexSBAPointXYZ*)optimizer.vertex(negVertexOffset + id*-1);
      |          ^~~~~
/home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1764:10: error: expected ‘)’ before ‘const’
[... 38 lines cut ...]
      |       ^
      |       pcl::fields::v
In file included from /usr/include/boost/preprocessor/variadic/elem.hpp:102,
                 from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:66,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.11/pcl/point_struct_traits.h:42,
                 from /usr/include/pcl-1.11/pcl/type_traits.h:40,
                 from /usr/include/pcl-1.11/pcl/memory.h:46,
                 from /usr/include/pcl-1.11/pcl/impl/point_types.hpp:41,
                 from /usr/include/pcl-1.11/pcl/point_types.h:349,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/Signature.h:32,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/Optimizer.h:37,
                 from /home/parker/build/rtabmap/corelib/src/../include/rtabmap/core/optimizer/OptimizerG2O.h:33,
                 from /home/parker/build/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:37:
/usr/include/pcl-1.11/pcl/impl/point_types.hpp:1893:1: note: ‘pcl::fields::v’ declared here
 1893 | POINT_CLOUD_REGISTER_POINT_STRUCT (pcl::_PointXYZHSV,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:839: corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:913: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

parkerlreed, 2021-01-11:
Full cmake output https://gist.githubusercontent.com/parkerlreed/610a043d0c6d7236dd95c69ca01ff8c2/raw/bc8170633e1be550860c25fd86275fdd6ce07a53/gistfile1.txt

parkerlreed, 2021-01-11:
g2o seems to be the only hang up. Explicitly disabling g2o, the rest compile and program runs (albeit not well without g2o support)

[image]

parkerlreed, 2021-01-11:
Seems to be working fine with the 2017 g2o. The wiki instructions just pull straight from master, so a note might need to be added.

(Was also missing libpointmatcher which I got installed before the final build)

[image]

closed 2021-01-11 by parkerlreed

matlabbe, 2021-01-11:
Normally g2o master would work, they changed the API 18 days ago: https://github.com/RainerKuemmerle/g2o/commit/0fcccb302787e70ff19f65e70fb103a1295b33a2

I'll see if I can detect which g2o version we are compiling to (they don't use versioning, always tricky to update for those api breaking issues)

parkerlreed, 2021-01-11:
@matlabbe Thanks. That worked great!

sumitsarkar1, 2021-01-20:
1.) Ubuntu 16
2.) Used the **eae5f2b** commit of rtabmap
3.) Used g2o master branch
4.) got the error : from /home/iobt/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:57:
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:27: error: ‘index_sequence’ is not a member of ‘std’
   struct HessianTupleType<std::index_sequence<Ints...>> {

5.) Had build RTabmap flawlessly in Nov 2020. Just changed Rtabmap for L515.
6.) All errors due to `base_fixed_sized_edge.h` of g2o. 
7.) Using CMAKE 3.18.5

commit eae5f2b4 (2021-01-11): fixed #674