introlab/rtabmap #675: Cannot use Apriltag as landmarks

Mingchi-Yeh, 2021-01-16:
I follow the other issue to set the landmarks : "In RTAB-Map, go to Preferences (advanced) -> Marker Detection panel, then check the checkbox to enable marker detection." 
But RTAB-Map can't choose the apriltag  to be landmarks, and RTAB-Map can't output the apriltag's coordinate axis either,as shown below.
[image]
[image]

Is there any support?
Thanks !

matlabbe, 2021-01-19:
OpenCV should be at least 3.4.2 to use Apriltag dictionary:
https://github.com/introlab/rtabmap/blob/5d777469ff46260991be81cdc4600f5bb8847365/corelib/src/MarkerDetector.cpp#L95-L99

If you are on Ubuntu 18.04 (opencv 3.2 by default), you would have to rebuild cv_bridge with newer OpenCV version. Otherwise, you could upgrade to Ubuntu 20.04, which uses opencv 4.2 by default: https://packages.ubuntu.com/search?keywords=libopencv-dev&searchon=names

Mingchi-Yeh, 2021-01-23:
Sorry, I am new to ROS. 
I use the ROS kinetic with ubuntu 16.
How do I change the OpenCV version of rtabmap to 3.4.2?
I tried to change the contents of this "cv_bridgeConfig.cmake" file to "/opt/ros/kinetic/share/cv_bridge/cmake".

```
if(NOT "include;/opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv "STREQUAL" ")
  set(cv_bridge_INCLUDE_DIRS "")
  set(_include_dirs "include;/opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv")
```

change to

```
if(NOT "include;/usr/local/include/opencv;/usr/local/include "STREQUAL" ")
  set(cv_bridge_INCLUDE_DIRS "")
  set(_include_dirs "include;/usr/local/include/opencv;/usr/local/include")
```

And
`set(libraries "cv_bridge;/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3.1;/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgproc3.so.3.3. 1;/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1")`

change to

`set(libraries "cv_bridge;/usr/local/lib/libopencv_core.so.3.4.2;/usr/local/lib/libopencv_imgproc.so.3.4.2;/usr/local/lib/libopencv_imgcodecs.so.3.4.2 ")`

And change "cv_bridge-extra.cmake" to

```
set(OpenCV_VERSION 3.4.2)
set(OpenCV_VERSION_MAJOR 3)
set(OpenCV_VERSION_MINOR 4)
set(OpenCV_VERSION_PATCH 2)
set(OpenCV_SHARED ON)
set(OpenCV_CONFIG_PATH /usr/local/share/OpenCV)
set(OpenCV_INSTALL_PATH /usr/loca)
set (OpenCV_LIB_COMPONENTS opencv_calib3d; opencv_core; opencv_dnn; opencv_features2d; opencv_flann; opencv_highgui; opencv_imgcodecs; opencv_imgproc; opencv_ml; opencv_objdetect; opencv_photo; opencv_shape; opencv_stitching; opencv_superres; opencv_video; opencv_videoio; opencv_videostab; opencv_viz; opencv_aruco; opencv_bgsegm; opencv_bioinspired; opencv_ccalib; opencv_cvv; opencv_datasets ; opencv_dpm; opencv_face; opencv_fuzzy; opencv_hdf; opencv_img_hash; opencv_line_descriptor; opencv_optflow; opencv_phase_unwrapping; opencv_plot; opencv_reg; opencv_rgbd; opencv_saliency; opencv_stereo; opencv_structured_light; opencv_surface_matching; opencv_text; opencv_tracking; opencv_xfeatures2d; opencv_ximgproc; opencv_xobjdetect; opencv_xphoto)
set(OpenCV_USE_MANGLED_PATHS TRUE)
set(OpenCV_MODULES_SUFFIX)
```
But it still can't build the package ,how do I use the OpenCV3.4.2?

matlabbe, 2021-01-24:
To avoid any conflicts, I suggest to uninstall libopencv from your computer. This will obviously uninstall many packages depending on OpenCV. Those packages will have to be rebuilt from source after you build/install new OpenCV version from source. This is why I suggested to upgrade to 20.04, you will avoid all those compilation issues.