introlab/rtabmap #679: Issues with locating rtabmap/MapCloud

s3668134, 2021-01-18:
Hello,

I've been trying to grab the map from the published rtabmap topic MapCloud and view it in rviz. However, I've been given this error

`The class required for this display, 'rtabmap_ros/MapCloud', could not be loaded. Error: Could not find library corresponding to plugin rtabmap_ros/MapCloud. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`

I know that I am missing the library package librtabmap_rviz_plugins.so which will hold the plugin but I don't know where to get it from. I would assume it already would have come from CMakeLists. However, I commented this line out [https://github.com/introlab/rtabmap_ros/blob/18ee9909fc2922e8015312b61be0dcab09c8309d/CMakeLists.txt#L29](url), I'm not sure if this could cause issues. I have also double checked whether I have sourced 'opt/ros/melodic/' first then the catkin workspace. 

If this information helps, here it is
Hardware: Jetson Nano
OS: Ubuntu 18.04 w/ ROS Melodic

How could I go about solving this issue?

matlabbe, 2021-01-19:
Well, if you commented `rviz` dependency, the MapCloud plugin will not be built: https://github.com/introlab/rtabmap_ros/blob/18ee9909fc2922e8015312b61be0dcab09c8309d/CMakeLists.txt#L263-L290

s3668134, 2021-01-19:
@matlabbe thank you for replying and offering your help. I assumed that would be the case.

I'm still a beginner with rtabmap, rviz and autonomous navigation as a whole and now I am unsure as to what I would need to do. I am trying to get a 3d map scan of the surroundings in real-time and be able to locate obstacles in the way. Since I can't use MapCloud, would I have to use PointCloud (like rtabmap/cloud_map) to do this? Or is MapCloud ideal? 

A more general question that I could ask is, could I grab a map from rtabmap and upload it to rviz in some form? Or could mapping be done in rviz alone? And how would I go about doing this?

matlabbe, 2021-01-19:
If you need to process the point cloud of the map, use /rtabmap/cloud_map. MapCloud is used more for visualization. IF you are moving in 2D, you may look for the 2D grid occupancy grid (/rtabmap/grid_map) that would be a more friendly format for navigation.

In RVIZ, add a PointCloud2 display, then subscribe it to /rtabmap/cloud_map topic to see the 3D cloud in rviz.

s3668134, 2021-01-20:
@matlabbe thank you for your help and clearing this up for me :)

closed 2021-01-20 by s3668134