introlab/rtabmap #680: Same problem again and again!!!

ManChrys, 2021-01-19:
Hi guys,

im using zed version :

[image]

rtabmap version : 

[image]

and on ZED section it says : NO

my cuda and drivers versions are : 

[image]
[image]

When i installed rtabmap on ZED Driver said : YES

[image]

On Detection --> stereo camera --> Zed_sdk is not available. I tried earlier versions on cuda 10.2 and had same problem. Maybe i cant understand something and always i mess up the connection between rtabmap and cuda. Hmmm i dont know im so confused. I hope you'll help me and open my eyes once again!!

Thanks in advance guys.

ManChrys, 2021-01-19:
@matlabbe you saved my life one more time!!!!!!!!!!!!! I was starting wrong rtabmap!!!!

Yeah im noob :D ty my friend :P Thank you for your patience :D

ManChrys, 2021-01-22:
Hello @matlabbe 
Well i found something else too.

Rtabmap works fine after your advise but when im trying to install rtabmap_ros and especially when im running the following command
$ sudo apt install ros-melodic-rtabmap ros-melodic-rtabmap-ros
rtabmap which i installed earlier from source doesnt work.
More specifically when i go to rtabmal/bin and run ./rtabmap the zed sdk is not available.
When i $sudo apt remove ros-melodic-rtabmap ros-melodic-rtabmap-ros it works fine again.

matlabbe, 2021-01-22:
You could try "ldd ./rtabmap" to see which libraries the linker is opening on runtime. Normally, it would load the libraries inside the same folder first.

ManChrys, 2021-01-25:
the result of the "ldd ./rtabmap" is the following. So its the same paths with rtabmap_ros? thats why this problem happens ?

geo@geo:~/rtabmap-0.20.8/bin$ ldd ./rtabmap
	linux-vdso.so.1 (0x00007ffd653de000)
	librtabmap_gui.so.0.20 => /opt/ros/melodic/lib/librtabmap_gui.so.0.20 (0x00007fdf9aa09000)
	librtabmap_core.so.0.20 => /opt/ros/melodic/lib/librtabmap_core.so.0.20 (0x00007fdf9a10a000)
	librtabmap_utilite.so.0.20 => /opt/ros/melodic/lib/librtabmap_utilite.so.0.20 (0x00007fdf99eec000)
	libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007fdf99ab1000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fdf9926a000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fdf98b1f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fdf98796000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fdf9857e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdf9818d000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fdf97f88000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdf97d69000)
	libpcl_common.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.8 (0x00007fdf97aad000)
	libpcl_io.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.8 (0x00007fdf975d2000)
	libpcl_search.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.8 (0x00007fdf971cb000)
	libpcl_surface.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8 (0x00007fdf96a56000)
	libpcl_visualization.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.8 (0x00007fdf966d8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fdf9633a000)
	libvtkIOMPIImage-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOMPIImage-6.3.so.6.3 (0x00007fdf96112000)
	libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 (0x00007fdf95f08000)
	libvtkRenderingMatplotlib-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.3.so.6.3 (0x00007fdf95cf7000)
	libvtkRenderingVolumeOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so.6.3 (0x00007fdf95a50000)
	libopencv_calib3d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2 (0x00007fdf95700000)
	libopencv_imgcodecs.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2 (0x00007fdf954c7000)
	libopencv_imgproc.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2 (0x00007fdf94f76000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007fdf94d07000)
	libvtkRenderingLOD-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.3.so.6.3 (0x00007fdf94aee000)
	libvtkGUISupportQt-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQt-6.3.so.6.3 (0x00007fdf948a4000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fdf9413b000)
	libvtkRenderingOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3 (0x00007fdf93c9b000)
	libvtkInteractionStyle-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.3.so.6.3 (0x00007fdf93a15000)
	libvtkRenderingAnnotation-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingAnnotation-6.3.so.6.3 (0x00007fdf936cb000)
	libvtkRenderingFreeType-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.3.so.6.3 (0x00007fdf93408000)
	libvtkRenderingCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingCore-6.3.so.6.3 (0x00007fdf92f88000)
	libvtkFiltersSources-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-6.3.so.6.3 (0x00007fdf92cc7000)
	libvtkFiltersGeneral-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersGeneral-6.3.so.6.3 (0x00007fdf92798000)
	libvtkCommonExecutionModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.3.so.6.3 (0x00007fdf924f1000)
	libvtkCommonDataModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.3.so.6.3 (0x00007fdf91f98000)
	libvtkCommonMath-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.3.so.6.3 (0x00007fdf91d78000)
	libvtkCommonCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.3.so.6.3 (0x00007fdf91858000)
	libopencv_stitching.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.3.2 (0x00007fdf915d6000)
	libopencv_aruco.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_aruco.so.3.2 (0x00007fdf9138e000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007fdf9112f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fdf90f12000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007fdf90c09000)
	libpcl_kdtree.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.8 (0x00007fdf908ce000)
	libpcl_filters.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.8 (0x00007fdf8feb4000)
	libpcl_features.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.8 (0x00007fdf8da09000)
	libpcl_sample_consensus.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.8 (0x00007fdf8cb62000)
	libpcl_segmentation.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.8 (0x00007fdf8bc3f000)
	libfreenect.so.0.5 => /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007fdf8ba2f000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007fdf8b7b9000)
	libg2o_core.so => /opt/ros/melodic/lib/libg2o_core.so (0x00007fdf8b561000)
	libg2o_types_slam2d.so => /opt/ros/melodic/lib/libg2o_types_slam2d.so (0x00007fdf8b30a000)
[... cut ...]
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007fdf58491000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007fdf58244000)
	libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007fdf5803d000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007fdf57e14000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007fdf57c05000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007fdf579bb000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fdf57783000)

matlabbe, 2021-01-25:
Yeah this is strange:
```
librtabmap_gui.so.0.20 => /opt/ros/melodic/lib/librtabmap_gui.so.0.20 (0x00007fdf9aa09000)
librtabmap_core.so.0.20 => /opt/ros/melodic/lib/librtabmap_core.so.0.20 (0x00007fdf9a10a000)
librtabmap_utilite.so.0.20 => /opt/ros/melodic/lib/librtabmap_utilite.so.0.20 (0x00007fdf99eec000)
```

It should load librtabmap_* libraries from the same folder, not the ros ones. You may try sudo ldconfig, or you could uninstall ros-melodic-rtabmap to make sure the right ones are used.

ManChrys, 2021-06-09:
@matlabbe 

thank you my friend. :D

closed 2021-06-09 by ManChrys