introlab/rtabmap #681: OptimizerG2O compile error

sumitsarkar1, 2021-01-20:
1.) Ubuntu 16, gcc 5.4.0 20160609
2.) Tried **#674** with commit **eae5f2b**. 
3.) Used g2o master 
3.) Got the error : `from /home/iobt/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:57:
/usr/local/include/g2o/core/base_fixed_sized_edge.h:174:27: error: ‘index_sequence’ is not a member of ‘std’
   struct HessianTupleType<std::index_sequence<Ints...>> {
`
All of the error are from `base_fixed_sized_edge.h` 
4.) Did build RTabmap flawlessly in Nov 2020. Updated due to L515
5.) Using CMAKE 3.18

matlabbe, 2021-01-20:
Try latest stable g2o release https://github.com/RainerKuemmerle/g2o/releases/tag/20201223_git
Looks like thy have broken again the API, I will recommend to stick on stable releases.

matlabbe, 2021-01-20:
Well, they upgraded to c++14 in this commit https://github.com/RainerKuemmerle/g2o/commit/4b04466e2e5c2da89744b92397c82d43ad00c140

I updated rtabmap in the commit above to use c++14 when g2o is detected.

closed 2021-01-20 by matlabbe