introlab/rtabmap #683: Error when make

Mingchi-Yeh, 2021-01-23:
Hi,
I'm trying to build the rtabmap by opencv3.4.2 , but I'm getting the following make error when building rtabmap from source on ubuntu 16 with ROS kinetic. 

cmake .. output is as follows :

```
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /usr/local/include;/usr/local/include/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/include/jsoncpp;/usr/include/hdf5/openmpi;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/python2.7;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found Freenect: /usr/include
-- Found DC1394: /usr/include/dc1394
-- Old g2o version detected with c++03 interface (config file: /opt/ros/kinetic/include/g2o/config.h).
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found RealSense2: 
-- Found octomap 1.8.1: /opt/ros/kinetic/include
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Architecture: x86_64
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.20.8
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++14
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
--   PCL_VERSION = 1.7.2
-- 
[... 153 lines cut ...]
main.cpp:(.text._ZN4YAML4NodeixIiEES0_RKT_[_ZN4YAML4NodeixIiEES0_RKT_]+0x3cf): undefined reference to `YAML::detail::node_data::insert_map_pair(YAML::detail::node&, YAML::detail::node&)'
main.cpp:(.text._ZN4YAML4NodeixIiEES0_RKT_[_ZN4YAML4NodeixIiEES0_RKT_]+0x3fb): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
CMakeFiles/euroc_dataset.dir/main.cpp.o: In function `main':
main.cpp:(.text.startup+0xb57): undefined reference to `YAML::LoadFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
main.cpp:(.text.startup+0x1687): undefined reference to `vtable for YAML::RepresentationException'
main.cpp:(.text.startup+0x16aa): undefined reference to `vtable for YAML::BadSubscript'
main.cpp:(.text.startup+0x16b1): undefined reference to `YAML::BadSubscript::~BadSubscript()'
main.cpp:(.text.startup+0x16bb): undefined reference to `typeinfo for YAML::BadSubscript'
main.cpp:(.text.startup+0x17fc): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
main.cpp:(.text.startup+0x1867): undefined reference to `YAML::detail::memory::create_node()'
main.cpp:(.text.startup+0x1934): undefined reference to `YAML::detail::memory_holder::merge(YAML::detail::memory_holder&)'
main.cpp:(.text.startup+0x196a): undefined reference to `YAML::detail::memory::create_node()'
main.cpp:(.text.startup+0x197f): undefined reference to `YAML::detail::node_data::insert_map_pair(YAML::detail::node&, YAML::detail::node&)'
CMakeFiles/euroc_dataset.dir/main.cpp.o:(.data.rel.ro._ZTIN4YAML18TypedBadConversionIiEE[_ZTIN4YAML18TypedBadConversionIiEE]+0x10): undefined reference to `typeinfo for YAML::BadConversion'
CMakeFiles/euroc_dataset.dir/main.cpp.o:(.data.rel.ro._ZTIN4YAML18TypedBadConversionIdEE[_ZTIN4YAML18TypedBadConversionIdEE]+0x10): undefined reference to `typeinfo for YAML::BadConversion'
CMakeFiles/euroc_dataset.dir/main.cpp.o:(.data.rel.ro._ZTIN4YAML18TypedBadConversionIfEE[_ZTIN4YAML18TypedBadConversionIfEE]+0x10): undefined reference to `typeinfo for YAML::BadConversion'
collect2: error: ld returned 1 exit status
tools/EurocDataset/CMakeFiles/euroc_dataset.dir/build.make:418: recipe for target '../bin/rtabmap-euroc_dataset' failed
make[2]: *** [../bin/rtabmap-euroc_dataset] Error 1
CMakeFiles/Makefile2:940: recipe for target 'tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all' failed
make[1]: *** [tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```
Can I get any help for this?

matlabbe, 2021-01-23:
It seems there is a conflict somehow with yaml-cpp library. 
https://github.com/introlab/rtabmap/blob/47e40ef34d3c1dc0e96409b6e3237f48ba836be4/tools/EurocDataset/CMakeLists.txt#L3

Otherwise, the yaml dependency is used only for the euroc command line tool. If you don't need it, you can comment this line:
https://github.com/introlab/rtabmap/blob/47e40ef34d3c1dc0e96409b6e3237f48ba836be4/tools/CMakeLists.txt#L9

Mingchi-Yeh, 2021-01-23:
It works !
Thank you for the help.