introlab/rtabmap #684: Cameras not detected

hai-h-nguyen, 2021-01-23:
Hello, 

I installed Rtabmap on Ubuntu 16.04 but Realsense cameras (Realsense D435i and T265) are not detected. More specifically, the option Realsense2 is grey when I chose the D435 or T265 option. I installed the SDK successfully because I can use Realsense-viewer to visualize the two cameras. What is wrong?

matlabbe, 2021-01-24:
rtabmap should be built from source to enable those options. Make sure when you do "cmake .." that "With RealSense2 = ON" so you know it will be enabled.

hai-h-nguyen, 2021-01-25:
Hi, I am able to make the option not grey. But when I click on it, nothing happens. Again, realsense-viewer still opens the cameras just fine.

jamesdegrandis, 2021-01-26:
I'm having a similar issue with the Realsense 2 on the L515.
I am getting "Camera Initialization Failue" and it says error.  "Config info not valid."
I don't think it's loading a valid YAML file for the camera.  I did update to the new SDK and firmware for the L515.
It does function in the Realsense viewer and I can capture data... but it seems like RTAB-MAP can't get a valid config on it.

Anyone have a manual config file I can put into camera_info that might work?

hai-h-nguyen, 2021-01-28:
@matlabbe 
I did not install some optional dependencies like OpenCV, g2o, GTSAM, or cvsba. Would it cause the software not to work like in my previous comment?

matlabbe, 2021-01-28:
@hai-h-nguyen can you set logger level to Debug (Preferences->Logging) to show more logs on the terminal?

hai-h-nguyen, 2021-01-28:
Hi, I just enable the logging function. I selected twice, once for D435i and then T265

[DEBUG] (2021-01-28 17:40:43.850) PreferencesDialog.cpp:1663::closeDialog() 
[DEBUG] (2021-01-28 17:41:00.631) PreferencesDialog.cpp:2621::writeSettings() _obsoletePanels=8 modified parameters=0
[DEBUG] (2021-01-28 17:41:00.631) MainWindow.cpp:4387::applyPrefSettings() 
[DEBUG] (2021-01-28 17:42:07.075) PreferencesDialog.cpp:2621::writeSettings() _obsoletePanels=8 modified parameters=0
[DEBUG] (2021-01-28 17:42:07.076) MainWindow.cpp:4387::applyPrefSettings()

closed 2021-01-28 by hai-h-nguyen