introlab/rtabmap #685: Error in `rtabmap': double free or corruption

sumitsarkar1, 2021-01-25:
1.) Ubuntu 16 LTS
2.) These were build from source : Eigen 3.3.7 , Ceres 1.14 , OpenCV 3.4.6, VTK 8.1, PCL 1.8, G2O (most recent on github) , GTSAM (most recent on github)  
3.) GTSAM , G2O, had options in CMake-gui to switch march native OFF and were turned off
4.) As per #343 I could not find march native OFF in PCL 1.8 CMakeLists.txt or in Eigen 3.3.7. Is PCL by default build with march native ON. if yes whats the variable that needed to be entered in cmake-gui (sorry I am novice to cmake !!)
5.) Otherwise with Eigen 3.2 , PCL 1.7 , GTSAM , all from `sudo apt install` and rest build from source works fine.
6.) Will PCL 1.7 and PCL 1.8 make a difference . @matlabbe One of your rtabmap odometry videos on youtube has PCL 1.8 and GTSAM as the optimizer choice in the parameter list !! Hence I ask.

Thank you

sumitsarkar1, 2021-01-25:
Solved ... build G2O and GTSAM with march native ON and then rebuild RTABMAP

closed 2021-01-25 by sumitsarkar1

matlabbe, 2021-01-25:
PCL 1.8 builds with `-march=native` by default. Only way to disable with PCL is to comment [this](https://github.com/PointCloudLibrary/pcl/blob/3fab805ae749dc5e1698720dbfad6d3dec4d6c15/cmake/pcl_find_sse.cmake#L7-L19). I think it is better to have it to get optimizations, but all libraries depending on Eigen should be built with it. G2O and GTSAM should be then also rebuilt from source with that option enabled. 

Note that PCL1.8 binaries installed using `apt get` don't have `-march=native`, so g2o and gtsam binaries can be used directly too.