introlab/rtabmap #687: RTABMAP using an RGB-D camera: [ERROR]Overwriting previous data! Make sure IMU is published faster than data rate.

PedromsLeal, 2021-02-15:
Hello. As you will infer, I'm new at ros and ubuntu.

Ubuntu 20.04.2 LTS
ROS Noetic
RTABMAP 0.20.7

I currently trying to use rtabmap to map a room. I have a Realsense L515, I'm currently using it as an RGB-D camera without point clouds to keep it simple and be able to follow the tutorials in ros site.

I run the rs_camera.launch.

The filter as in the ros tutorial for the D435i:
rosrun imu_filter_madgwick imu_filter_node \
    _use_mag:=false \
    _publish_tf:=false \
    _world_frame:="enu" \
    /imu/data_raw:=/camera/imu \
    /imu/data:=/rtabmap/imu

And the rtabmap.launch like:
roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=/camera/depth/image_rect_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=true \
    wait_imu_to_init:=true \
    imu_topic:=/rtabmap/imu \
    rviz:=true \
    rtabmapviz:=false \
    rgbd_sync.=true \
    approx_rgbd_sync = true 

Before I had set those last 2 lines (    rgbd_sync.=true \  approx_rgbd_sync = true) everytime I attempted to run rtabmap these errors appeared immediately:

[ERROR] [1613430364.289239210]: Overwriting previous data! Make sure IMU is published faster than data rate. (last image stamp buffered=1613430364.188161 and new one is 1613430364.250351, last imu stamp received=1613430364.185519)
+ 
[ WARN] [1613430750.278127668]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=100).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /camera/depth/image_rect_raw \
   /camera/color/camera_info \
   /rtabmap/odom_info

I then realized that for some reason rgbd_odometry wasn't working because /rtabmap/odom wasn't being published.
So I set rgbd_sync.=true \  approx_rgbd_sync = true in rtabmap.launch.

Now it is able to work for around 10 seconds, where it prints

[ INFO] [1613430955.887916829]: Odom: quality=319, std dev=0.003297m|0.056744rad, update time=0.035832s
[ INFO] [1613430955.938093264]: Odom: quality=301, std dev=0.003539m|0.068844rad, update time=0.036089s
[ INFO] [1613430956.004733540]: Odom: quality=305, std dev=0.002702m|0.044810rad, update time=0.032671s
[ INFO] [1613430956.104110124]: Odom: quality=295, std dev=0.002502m|0.038960rad, update time=0.068838s
[ INFO] [1613430956.142271883]: Odom: quality=334, std dev=0.002724m|0.044810rad, update time=0.036807s

However, after a bit, the following error appears :

[ERROR] [1613430963.780072744]: Overwriting previous data! Make sure IMU is published faster than data rate. (last image stamp buffered=1613430963.680351 and new one is 1613430963.744697, last imu stamp received=1613430963.670266)
[ERROR] [1613430963.826646038]: Overwriting previous data! Make sure IMU is published faster than data rate. (last image stamp buffered=1613430963.744697 and new one is 1613430963.810415, last imu stamp received=1613430963.740007)
[ERROR] [1613430963.887505296]: Overwriting previous data! Make sure IMU is published faster than data rate. (last image stamp buffered=1613430963.810415 and new one is 1613430963.870398, last imu stamp received=1613430963.804767)
[ERROR] [1613430963.955757260]: Overwriting previous data! Make sure IMU is published faster than data rate. (last image stamp buffered=1613430963.870398 and new one is 1613430963.936713, last imu stamp received=1613430963.869526)

First it appears now and then, but after 5-10 seconds it no longer shows the previous [INFO] lines and only shows the error.
While this happened I ran rostopic hz /rtabmap/odom, and it showed that while it starts at around 17 per second, as the errors appear it's rate falls to 0.

As i've said it takes around 10 seconds until it completely stops working and it is flooded with the error, something I noticed is that it's around 10 seconds from when I run the rs_camera.launch + madwick filter.
So if I wait those ~10 seconds before I launch rtabmap, when it starts it is already showing the error.

I cannot find any mention of this error anywhere, any help would be appreciated.
@matlabbe   
Sorry for the mention! Just wanted to make sure this was seen.

Just the launch files:
[rs_camera.zip](https://github.com/introlab/rtabmap/files/5984629/rs_camera.zip)
[rtabmap.zip](https://github.com/introlab/rtabmap/files/5984647/rtabmap.zip)

rlabs-oss, 2021-02-16:
Hi,
    Is mapping using ROS essential ? Have you tried the standalone version and the L515? One thing to check is make sure the [latest](https://github.com/IntelRealSense/librealsense/releases/tag/v2.41.0) realsense sdk is in use and the camera firmware is the latest. Global time sync used to cause problems in the past but testing it today on and off didn't seem to make much difference in rtab-map standalone.

PedromsLeal, 2021-02-19:
I will test it soon, I have been having trouble to use the standalone because it does not detect realsense2 when I'm am trying to select the L515 camera, even though I can perfectly use it in ROS.

matlabbe, 2021-02-21:
>  it does not detect realsense2 when I'm am trying to select the L515 camera

rtabmap should be built from source to have the menu option enabled.

For the imu warning, when it happens, check if the imu data is actually published or not. How the realsense2 node is started? what is `unite_imu_method` set to?