introlab/rtabmap #690: 3D Map is not loaded

Everest829, 2021-02-22:
hareware：jetson Xavier NX
OS：Ubuntu 18.04，ROS Melodic 
Environment:CUDA 10.2.89 ; Opencv 4.4.0;ZED 2;ZED-SDK：ZED_SDK_Tegra_JP45_v3.4.0.run

I can't view the 3D Map generated when I launch the zed2.launch. The commands I executed are listed as follows：
$roslaunch zed_wrapper zed2.launch camera_model:=zed2；
$roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/zed2/zed_node/rgb/image_rect_color depth_topic:=/zed2/zed_node/depth/depth_registered camera_info_topic:=/zed2/zed_node/rgb/camera_info frame_id:=base_link approx_sync:=false visual_odometry:=false odom_topic:=/zed2/zed_node/odom.
Finally,the result I got is shown as in the figure：
[image]

matlabbe, 2021-02-22:
Unfortunately, this is an issue with the jetpack with incompatible VTK versus Qt libraries. See https://github.com/introlab/rtabmap/issues/427

I recommend to use RVIZ instead for visualization on the jetson. If you are on many computers and you want to use rtabmapviz, you could launch rtabmapviz on a laptop.

Everest829, 2021-02-23:
I have followed the method #427 ,but I got a new question when I input the shell command _cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=4 -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release ..
I have remove all Qt5 related vtk libraries installed in `/usr/lib/aarch64-linux-gnu`，and I install qt4
[image]

matlabbe, 2021-02-26:
Which VTK version is used? PCL/Rtabmap won't build with VTK>=9. It looks like cmake is looking for QtWebkit, which is not installed.

Everest829, 2021-02-26:
I built vtk 6.3.0 from source，but I couldn‘t make it.The Qt version I've installed is Qt5.I followed the method you recommended，then I uninstalled Qt5  and I installed Qt4 successfully。In Part2，I can not execut the last step Cmake. Error occured，
[image].
So The place occured error is commented out 
[image]
Finally，I executed '_cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=4 -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release ..',I got the result
[image]

matlabbe, 2021-02-28:
Your error2.png failed to be uploaded

matlabbe, 2021-11-03:
Another solution is to use the docker image, see https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-desktop-ubuntu-1804-2004 (updated from this issue https://github.com/introlab/rtabmap/issues/776), the 3D view is working properly.

YXH11111, 2023-04-10:
> 
**Have you ever been in this situation? I follow(https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-desktop-ubuntu-1804-2004) .**

root@robot-desktop:~# mkdir rtanmap3d
root@robot-desktop:~# cd rtabmap3d
root@robot-desktop:~/rtabmap3d# touch Dockerfile
root@robot-desktop:~/rtabmap3d# gedit Dockerfile
root@robot-desktop:~/rtabmap3d# docker build -t rtabmap3d .
Sending build context to Docker daemon  2.048kB
Step 1/5 : FROM introlab3it/rtabmap:bionic
 ---> 1af0d1efdfa7
Step 2/5 : ENV NVIDIA_VISIBLE_DEVICES     ${NVIDIA_VISIBLE_DEVICES:-all}
 ---> Running in efe9d8ec9c57
Removing intermediate container efe9d8ec9c57
 ---> 4ccac62f37c3
Step 3/5 : ENV NVIDIA_DRIVER_CAPABILITIES     ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics
 ---> Running in b7548ffb38c0
Removing intermediate container b7548ffb38c0
 ---> ca036938154b
Step 4/5 : RUN mkdir -p /root/Documents/RTAB-Map
 ---> Running in df2799737116
Removing intermediate container df2799737116
 ---> 4e7e43c6b258
Step 5/5 : ENV LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/aarch64-linux-gnu/tegra
 ---> Running in f0149eabcc12
Removing intermediate container f0149eabcc12
 ---> 0ae38ee7acf8
Successfully built 0ae38ee7acf8
Successfully tagged rtabmap3d:latest
root@robot-desktop:~/rtabmap3d#  export XAUTH=/tmp/.docker.xauth
root@robot-desktop:~/rtabmap3d#  touch $XAUTH
root@robot-desktop:~/rtabmap3d#  xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
root@robot-desktop:~/rtabmap3d# docker run -it --rm \
>    --privileged \
>    --env="DISPLAY=$DISPLAY" \
>    --env="QT_X11_NO_MITSHM=1" \
>    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
>    --env="XAUTHORITY=$XAUTH" \
>    --volume="$XAUTH:$XAUTH" \
>    --runtime=nvidia \
>    --network host \
>    -v ~/Documents/RTAB-Map:/root/Documents/RTAB-Map \
>    rtabmap3d \
>    rtabmap
docker: Error response from daemon: OCI runtime create failed: failed to create NVIDIA Container Runtime: failed to construct OCI spec modifier: failed to construct discoverer: failed to create Xorg discoverer: failed to locate libcuda.so: pattern libcuda.so.*.*.* not found: unknown.

matlabbe, 2023-04-16:
Which computer are you using? the docker error contain cuda not found libraries, this is not rtabmap related.

YXH11111, 2023-04-17:
Thanks,I will check it out.

杨新慧
***@***.***

杭州电子科技大学