introlab/rtabmap #691: No space left on hard drive when using Zed2 stereo camera in uvc mode

matlabbe, 2021-02-26:
System: Ubuntu 18.04

When launching StereoCameraVideo with a zed2 camera, the hard drive is filling fast until there is no more space:
```bash
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.7G     0  7.7G   0% /dev
tmpfs           1.6G   11M  1.6G   1% /run
/dev/nvme0n1p7  288G  288G     0 100% /
tmpfs           7.7G  172K  7.7G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
...
```

The problem can be fixed following this post by deleting `/var/log/uvcdynctrl-udev.log` and setting `uvcdynctrl` debug flag to 0: https://askubuntu.com/questions/177312/filesystem-filling-up-due-to-large-uvcydnctrl-udev-log-file

On my computer the log file increased to 45 GB to fill the hard drive:
ls -l /var/log
...
-rw-r--r--  1 root              root            45229785088 Feb 25 22:06 uvcdynctrl-udev.log
...

closed 2021-02-26 by matlabbe

Myzhar, 2021-02-26:
@matlabbe have you tried to integrate the ZED Open Capture driver which does not require CUDA to run?
It's a good replacement for the pure UVC capture
It allows you to set camera control values and to retrieve sensors data from ZED-mini and ZED2

matlabbe, 2021-02-26:
Great! it looks super nice as we can also get IMU data along factory calibration: https://github.com/stereolabs/zed-open-capture/blob/master/examples/zed_oc_sync_example.cpp

I'll take some time to add this option for zed cameras.