introlab/rtabmap #694: Intel Realsense is greyed out

Zumbalamambo, 2021-03-03:
I have installed RTABMAP through ROS-Noetic. The version of RTABMAP that I have installed is v20.7. For some reason, the intel realsense is greyed out both in the setting as well in the RGB-D Menu. Do I need to install any other dependencies?

matlabbe, 2021-03-03:
If you[ use realsense with realsense ros package](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping), you don't need to enable it in rtabmapviz.

However, if you want to use rtabmap standalone version with realsense driver, you have to rebuild [rtabmap from source](https://github.com/introlab/rtabmap_ros#build-from-source). Make sure after you do `cmake ..`, you see `With RealSense2 = YES`. Make sure to uninstall rtabmap binaries to avoid possible mismatch library version issues afterwards.
```bash
$ sudo apt remove ros-noetic-rtabmap

$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake .. 
# Verify "--   With RealSense2           = YES (License: Apache-2)"
$ make
$ sudo make install
```

Zumbalamambo, 2021-03-03:
@matlabbe thank you! Is it possible to install rtabmap with realsense in nano kit?

matlabbe, 2021-03-05:
The standalone won't work out of the box (see this [long issue](https://github.com/introlab/rtabmap/issues/427)). However, if you want to use RVIZ for visualization, the rtabmap binaries will work out of the box.

Zumbalamambo, 2021-03-05:
thank you! just one last question, if I install RTABMap in windows, do I still have to build from source to be able to use rtabmap with realsense or does it comes bundled with the realsense libraries?

matlabbe, 2021-03-05:
The binaries are bundled with realsense2 support, see https://github.com/introlab/rtabmap/releases

closed 2021-03-06 by Zumbalamambo