introlab/rtabmap #700: L515 with T265

peta-peta, 2021-03-17:
I try to make 3D models of rooms for some time now using L515 and D455. Over the last year the scans become much more reliable and rtabmap has improved a lot.

My experience: I appreciate the accuracy of the L515 over the d455. Straight walls are much better captured with the L515 and the exported 3D maps exhibit more details. With the d455 the scans look "noisy" but mapping works much better than with the L515.
I wonder if it is worth to buy the T265 and combine it with L515 to perform odometry. My expectation is that mapping will work much better than with the L515 only. Anyone of you tried L515 with T265 and can confirm my expectation?

rlabs-oss, 2021-03-17:
Hi Peta, 
            I made a quick test but did not fully correct the exact L515 pose relative to the T265. My T265 is above a mid 40 lidar. I placed the L515 in front of the mid40 and below the T265. This corresponded to an X offset around 40mm, Y 50, Z 30, assuming that's 0.04 0.05 0.03 0 0 0? Hold this contraption steady I made a quick scan and found the result was poor, overlapping areas.  The L515 alone was clean. It maybe different with a custom bracket? But based on my experience of the T265 and mid40 its not a brilliant solution for hand held use.  

[image]

peta-peta, 2021-03-17:
Thank you for you quick response.
Are you satisfied with the l515 mapping results? Do you mind to share your setup?

rlabs-oss, 2021-03-17:
I am very satisfied with RTAB-Map and the results that can be obtained. Its an open source integration / research tool rather than polished product. Always good to be mindful of this and those that contribute towards the effort. If code is not being contributed documentation and also configurations or how-to guides maybe most welcome. I've thought about this for the L515. 

The setup or RTAB-Map configuration files vary depending on the machine I use. I normally start from default. For laptop / interior mapping, Memory/Image pre decimation (Mem/ImagePreDecimation) 2 can help (or hinder if features are scarce). The madgewick filter is enabled by default now. 3D rendering at 4. Walking backwards over an already scanned area rather than turning around and walking forwards helps also. After mapping, Edit/update cache, followed by tools post processing to fix any loop closure issues. Normally the saved .db I will load again using the source = database and process again on a different machine than it was captured - depends on what I'm doing.  Fixing the L515 exposure via an exported json configuration from realsense-viewer may also help with patchy scans due to light changes. Thus load that via D400 Series Visual presets option in the RealSense2 area/below the dual mode extrinsics.
Ive probably not covered all cases here but hope this helps.

matlabbe, 2021-03-19:
A small note on L515+T265, I've seen in some datasets that T265 translation has not always the same scale than an odometry computed with a lidar at the same time. For example, the T265 will estimate a displacement of 1 meter while if we look at lidar data it should have move 1.1 meters. I don't know if we can calibrate again the T265 in case the factory calibration has some scaling issues. So when combined with L515 giving some depth values, if the trajectory has not the same scale, the walls may don't overlap perfectly with T265 odometry. I don't know if there is someone that actually measured the real distance against T265 computed distance while walking 10 meters straight for example. Without calibration, a fix could be to scale the odometry pose based on the ratio of distance measured in reality.

L515 RGB-D visual odometry is very poor because of the high motion blur in RGB camera. You can do RGB-D using only IR camera, I saw some improvements, though very sensitive to external IR in the environment. There is also a possibility to ICP odometry instead, like in this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Kinect-For-Azure-L515-ICP-lighting-invariant-mapping-td7187.html).

cheers,
Mathieu

peta-peta, 2021-03-24:
I tried your ICP odometry using config_l515_icp.ini. Odometry and mapping is very good :) Is it also possible to save  images with the RGB camera at the same time? I would like to use the pictures to texture the mesh.

rlabs-oss, 2021-03-25:
In Source deselect the IR mode - however move the camera slowly since the issues with motion blur may be back as mentioned earlier by Mathieu. - Realsense viewer may also be used to set the exposure, export the configuration/json file which if specified in the D400 series presets may help reduce the patchwork effect for overlapping sections.
[image]

matlabbe, 2021-03-28:
It is possible to do icp odometry while recording RGB images. Starting from config_l515_icp.ini, uncheck IR mode.

peta-peta, 2021-03-28:
Thanks for your hints @rlabs-oss and @matlabbe. It works very well. As you recommended, I started from loading  config_l515_icp.ini and then unchecked IR mode.
Is there any literature on ICP that you can recommend to read to better understand what happens behind the scene?

rlabs-oss, 2021-03-28:
Cyrill Stachniss recent video series on ICP https://www.youtube.com/watch?v=dhzLQfDBx2Q 
Favorite beverage and snacks are mandatory :)

sumitsarkar1, 2021-04-02:
@peta-peta any links to config_l515_icp.ini or if you can share please

peta-peta, 2021-04-02:
> @peta-peta any links to config_l515_icp.ini or if you can share please

peta-peta, 2021-04-02:
> @peta-peta any links to config_l515_icp.ini or if you can share please

Her you go: https://github.com/introlab/rtabmap/releases/download/0.20.8/config_l515_icp.ini
Enjoy :)

closed 2021-04-02 by peta-peta