introlab/rtabmap #701: Extract logged node covariance

eliabntt, 2021-03-19:
Good morning, I've a quick question. Starting from a database is it possible to get a text/csv file with the covariance (linear/angular) of all nodes? I've seen there's the possibility to plot it with Memory/Odometry_variance_ang/ but cannot find a way to directly save it as log.

eliabntt, 2021-03-19:
Side question, when I export poses (through rtabmap-databaseViewer -> export poses) the link covariance that I have in the log (for example in g2o) is the one optimized after LC or the raw one?

matlabbe, 2021-05-02:
I think you already answered your first question. The g2o export contains raw covariance (in information matrix format). In Graph view, if you move the slider left to 0, you will export odometry poses. The slider shows the graph and each iteration step of g2o or gtsam (0 meaning no optimization). When you export poses, the current graph selected by the slider is exported (covariances are all the same (raw ones) independently of the iteration chosen).

eliabntt, 2021-05-03:
aren't those link based [so information matrix of the link from node x to node y] and not node's ones [so covariance of the odometry at the time of the node addition/of the single node after the optimization]?

matlabbe, 2021-05-04:
The information matrix between on link between node x and node y would be the odometry twist covariance matrix received when node y has been created.