introlab/rtabmap #702: .bag file source

FLS-BP-US, 2021-03-22:
Is it possible to add a .bag file reader as a source?  I want to re-run the same data set with different settings to find an optimal parameters.

matlabbe, 2021-03-27:
The problem with bags is that they would not have standard topic names and topic types. One could one to do stereo mapping, while other RGB-D mapping, or even use 2D laser scans and 3D point clouds. The synchronization of topics can also be an issue. A workaround is to create a source Database from the rosbag with [data_recorder.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch) (see the file for arguments for which topics you want to record):

```bash
roscore
rosparam set use_sim_time true
roslaunch rtabmap_ros data_recorder.launch
rosbag play --clock input.bag
```
The `output.db` database generated could be then used as source in rtabmap standalone.

FLS-BP-US, 2021-04-01:
I will give that a try.  Thanks

closed 2021-04-01 by FLS-BP-US