introlab/rtabmap #706: Error running rtabmap_ros

NunoGuedesFollow, 2021-04-01:
Hi, I'm running Ubuntu 18.04 with rtabmap 0.20.7 with PCL 1.8.1, VTK 6.3.0, OpenCV 4.5.1 with xFeatures2d and without nonfree.
Also it was built with default g2o, GTSAM from source with flag -DGTSAM_USE_SYSTEM_EIGEN=ON and libpointmatcher from source. Also I installed eigen from source without march-native.

The error that appears is the following (using gdb)
[image]

Can anyone help?

matlabbe, 2021-04-01:
If you do `bt` in gdb terminal, what does it show?

NunoGuedesFollow, 2021-04-01:
> If you do `bt` in gdb terminal, what does it show?

[image]

matlabbe, 2021-04-01:
It seems a compilation problem. Can you show the output of of cmake of rtabmap library?  (rtabmap 0.20.7 is the binaries or from source?)

Look where librtabmap_ros.so is built (catkin_ws/devel/lib or /opt/ros/[rosdistro]/lib) and do ldd:
```
ldd ./librtabmap_ros.so
```

NunoGuedesFollow, 2021-04-01:
RTabmap is from source.
Here is the output of cmake ..

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /opt/OpenCV4/include/opencv4
-- Found PCL: /usr/include/pcl-1.8;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.3;/usr/include/freetype2;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include;/usr/lib/x86_64-linux-gnu/openmpi/include;/usr/include/python2.7;/usr/include/x86_64-linux-gnu;/usr/include/hdf5/openmpi;/usr/include/libxml2;/usr/include/jsoncpp;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found Torch7 in /home/nunoguedes/Libraries/torch/install
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /opt/ros/melodic/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:404 (FIND_PACKAGE)

-- Found g2o: /opt/ros/melodic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   filesystem
--   system
--   program_options
--   date_time
--   chrono
--   atomic
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/local/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:10 (find_package)
  CMakeLists.txt:430 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
[... 83 lines cut ...]
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (VINS-Fusion not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nunoguedes/Libraries/rtabmap/build

Also the ldd didn't solve this problem.

matlabbe, 2021-04-02:
This is strange that the following message is not completely shown in the log:
https://github.com/introlab/rtabmap/blob/d8ebbc2645b68b4735d030ff2ceba162499918bc/corelib/src/RegistrationIcp.cpp#L548

Make sure rtabmap binaries are uninstalled before doing the catkin_make. 
```
sudo apt remove ros-melodic-rtabmap
```

The malloc error happens in a std::map. Can you test a build without GTSAM? 
```
cd rtabmap/build
cmake -DWITH_GTSAM=OFF ..
make -j6
make install
```

NunoGuedesFollow, 2021-04-05:
I also noticed that print that is incomplete...
So, i dont have any rtabmap binaries installed (didn't have before also) and i tried compiled with GTSAM off and ran again and got the following error:
[image]

matlabbe, 2021-05-02:
To debug this, I often try building with minimal dependencies and also avoid dependencies built from source (like your OpenCV). You could open with cmake-gui and uncheck all `WITH_****`. Then re-add incrementally each dependency one by one until it crashes again. If rtabmap is still crashing wihout any dependencies and when also using opencv binaries, then I could try to reproduce the bug in a docker.

NunoGuedesFollow, 2021-05-03:
Hi!
Thanks for the help, but the problem was the source eigen version I was using to build the packages. 
Once I used the default version, the compilation worked.

closed 2021-05-03 by NunoGuedesFollow