introlab/rtabmap #709: Is CUDA compulsory for SuperPoint

sumitsarkar1, 2021-04-15:
Tried configuring SuperPoint with RTABMAP as per [#531](https://github.com/introlab/rtabmap/issues/531#issuecomment-615221478). I get the following error :
```
VTK_RENDERING_BACKEND=OpenGL
CUDA_TOOLKIT_ROOT_DIR not found or specified
Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
CMake Warning at /home/robovision/libtorch/share/cmake/Caffe2/public/cuda.cmake:31 (message):
  Caffe2: CUDA cannot be found.  Depending on whether you are building Caffe2
  or a Caffe2 dependent library, the next warning / error will give you more
  info.
Call Stack (most recent call first):
  /home/robovision/libtorch/share/cmake/Caffe2/Caffe2Config.cmake:88 (include)
  /home/robovision/libtorch/share/cmake/Torch/TorchConfig.cmake:40 (find_package)
  CMakeLists.txt:332 (FIND_PACKAGE)

CMake Error at /home/robovision/libtorch/share/cmake/Caffe2/Caffe2Config.cmake:90 (message):
  Your installed Caffe2 version uses CUDA but I cannot find the CUDA
  libraries.  Please set the proper CUDA prefixes and / or install CUDA.
Call Stack (most recent call first):
  /home/robovision/libtorch/share/cmake/Torch/TorchConfig.cmake:40 (find_package)
  CMakeLists.txt:332 (FIND_PACKAGE) 
```

matlabbe, 2021-05-04:
Is cuda in your PATH?
https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#environment-setup

sumitsarkar1, 2021-05-05:
I dont have CUDA as I dont have GPU...so does that mean I wont be able to use SuperPoint on only CPU platforms

matlabbe, 2021-05-05:
You should be able to use with only CPU. However, you need to install/build libtorch without CUDA depedency.

sumitsarkar1, 2021-05-05:
Even for loop closure at 1 Hz?

closed 2021-05-05 by sumitsarkar1

matlabbe, 2021-05-05:
1 hz maybe, depending on your cpu.