introlab/rtabmap #720: Ros docker introlab3it/rtabmap_ros:melodic not built with multiRGBD

dagata-mining, 2021-05-12:
When I run this command
`sudo docker run -it --rm  --env ROS_MASTER_URI=http://172.17.0.1:11311 --env ROS_IP=172.17.0.2  -v ~/.ros:/root   introlab3it/rtabmap_ros:melodic roslaunch /root/my_launch_file.launch`

I get the error :

` [FATAL] [1620789173.224815374]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)  `

So I'm wondering how to have an image with the RTABMAP_SYNC_MULTI_RGBD  ON

Also if Im building with Introlab3it/rtabmap_ros:melodic-latest I get the error:

`[rtabmap/rtabmap-37] process has died [pid 366, exit code -11, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmap --delete_db_on_start scan_cloud:=assembled_cloud imu:=/sensor_t265_1/imu/data __name:=rtabmap __log:=/root/.ros/log/0c70268a-b2ce-11eb-b47c-d540a5fe237a/rtabmap-rtabmap-37.log].
log file: /root/.ros/log/0c70268a-b2ce-11eb-b47c-d540a5fe237a/rtabmap-rtabmap-37*.log
`

and

`[rtabmap/icp_odometry-36] process has died [pid 327, exit code -11, cmd /opt/ros/melodic/lib/rtabmap_ros/icp_odometry scan_cloud:=combined_cloud __name:=icp_odometry __log:=/root/.ros/log/0c70268a-b2ce-11eb-b47c-d540a5fe237a/rtabmap-icp_odometry-36.log].
log file: /root/.ros/log/0c70268a-b2ce-11eb-b47c-d540a5fe237a/rtabmap-icp_odometry-36*.log
`

matlabbe, 2021-05-12:
I tested with that option enabled, see [commit](https://github.com/introlab/rtabmap_ros/commit/0d693bc1b6067be26f3baa244f7cff471c495aa5) above.

```bash
$ cd rtabmap_ros/docker/melodic/latest

$ docker build -t introlab3it/rtabmap_ros:melodic-latest .

$ roscore

$ docker run -it --rm \
 --env ROS_MASTER_URI=http://172.17.0.1:11311 --env ROS_IP=172.17.0.2 \
 -v ~/.ros:/root  \
 introlab3it/rtabmap_ros:melodic-latest \
 rosrun rtabmap_ros rtabmap --delete_db_on_start _subscribe_rgbd:=true _rgbd_cameras:=4 _database_path:=/root/rtabmap.db
```

Log: 
```
[ INFO] [1620821203.831504560]: Starting node...
[ INFO] [1620821203.853207616]: Initializing nodelet with 12 worker threads.
[ INFO] [1620821203.994655805]: /rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1620821203.994716100]: /rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1620821203.994730999]: /rtabmap(maps): map_cleanup                = true
[ INFO] [1620821203.994755935]: /rtabmap(maps): map_always_update          = false
[ INFO] [1620821203.994788533]: /rtabmap(maps): map_empty_ray_tracing      = true
[ INFO] [1620821203.994816704]: /rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1620821203.994829266]: /rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1620821203.994857636]: /rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1620821203.995557597]: /rtabmap(maps): octomap_tree_depth         = 16
[ INFO] [1620821204.009472039]: rtabmap: frame_id      = base_link
[ INFO] [1620821204.009514626]: rtabmap: map_frame_id  = map
[ INFO] [1620821204.009528259]: rtabmap: use_action_for_goal  = false
[ INFO] [1620821204.009569197]: rtabmap: tf_delay      = 0.050000
[ INFO] [1620821204.009602249]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1620821204.009642740]: rtabmap: odom_sensor_sync   = false
[ INFO] [1620821204.009978097]: rtabmap: gen_scan  = false
[ INFO] [1620821204.010011200]: rtabmap: gen_depth  = false
[ WARN] [1620821204.337084349]: Setting "Vis/EstimationType" parameter to 0 (1 is not supported for multi-cameras) as "subscribe_rgbd" is true and "rgbd_cameras">1. Set "Vis/EstimationType" to 0 to suppress this warning.
[ INFO] [1620821204.337744438]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1620821204.338069592]: rtabmap: Deleted database "/root/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1620821204.338133826]: rtabmap: Using database from "/root/rtabmap.db" (0 MB).
[ INFO] [1620821204.680743528]: rtabmap: Database version = "0.20.10".
[ INFO] [1620821204.680887490]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[ WARN] [1620821204.762883668]: rtabmap: Parameters subscribe_depth and subscribe_rgbd cannot be true at the same time. Parameter subscribe_depth is set to false.
[ INFO] [1620821204.769578060]: /rtabmap: subscribe_depth = false
[ INFO] [1620821204.769665575]: /rtabmap: subscribe_rgb = false
[ INFO] [1620821204.769716337]: /rtabmap: subscribe_stereo = false
[ INFO] [1620821204.769838603]: /rtabmap: subscribe_rgbd = true (rgbd_cameras=4)
[ INFO] [1620821204.769924432]: /rtabmap: subscribe_odom_info = false
[ INFO] [1620821204.770030049]: /rtabmap: subscribe_user_data = false
[ INFO] [1620821204.770113720]: /rtabmap: subscribe_scan = false
[ INFO] [1620821204.770228536]: /rtabmap: subscribe_scan_cloud = false
[ INFO] [1620821204.770394362]: /rtabmap: subscribe_scan_descriptor = false
[ INFO] [1620821204.770505674]: /rtabmap: queue_size    = 10
[ INFO] [1620821204.770612050]: /rtabmap: approx_sync   = true
[ INFO] [1620821204.770832676]: Setup rgbd4 callback
[ INFO] [1620821204.790409436]: 
/rtabmap subscribed to (approx sync):
   /odom \
   /rgbd_image0 \
   /rgbd_image1 \
   /rgbd_image2 \
   /rgbd_image3
[ INFO] [1620821204.875023314]: rtabmap 0.20.10 started...
```

dagata-mining, 2021-05-12:
Thanks, your build did work

```
docker run -it --rm \
 --env ROS_MASTER_URI=http://172.17.0.1:11311 --env ROS_IP=172.17.0.2 \
 -v ~/.ros:/root  \
 introlab3it/rtabmap_ros:melodic-latest \
 rosrun rtabmap_ros rtabmap --delete_db_on_start  _subscribe_rgbd:=true _rgbd_cameras:=4 _database_path:=/root/rtabmap.db
```

However I can't manage to run the docker with a single launch file and I do set the   delete_db_on_start  _subscribe_rgbd:=true _rgbd_cameras:=4 _database_path:=/root/rtabmap.db in my launch file

Is it possible to run it from a single launch file?

matlabbe, 2021-05-12:
Yes, you should be able to run a launch file. In your example, if your launch file `my_launch_file.launch` is in `~/.ros` directory on host computer, then from inside docker it would be `roslaunch my_launch_file.launch` or `roslaunch /root/my_launch_file.launch` (normally docker would run from `/root` workdir though)

dagata-mining, 2021-05-12:
Thanks I'm able to launch the file, however it crashes at 2 different places 

`[rtabmap/icp_odometry-36] process has died [pid 328, exit code -11, cmd /opt/ros/melodic/lib/rtabmap_ros/icp_odometry scan_cloud:=combined_cloud __name:=icp_odometry __log:=/root/.ros/log/939aaba8-b31e-11eb-b47c-d540a5fe237a/rtabmap-icp_odometry-36.log].
log file: /root/.ros/log/939aaba8-b31e-11eb-b47c-d540a5fe237a/rtabmap-icp_odometry-36*.log
`
and

`[rtabmap/rtabmap-37] process has died [pid 367, exit code -11, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmap --delete_db_on_start scan_cloud:=assembled_cloud imu:=/sensor_t265_1/imu/data __name:=rtabmap __log:=/root/.ros/log/aa760e98-b33d-11eb-b47c-d540a5fe237a/rtabmap-rtabmap-37.log].
log file: /root/.ros/log/aa760e98-b33d-11eb-b47c-d540a5fe237a/rtabmap-rtabmap-37*.log
`

dagata-mining, 2021-05-12:
Unfortunately it did not solve the problem

matlabbe, 2021-05-12:
If you do:
```bash
$ docker pull introlab3it/rtabmap_ros:melodic-latest
$ roscore
$ docker run -it --rm \
 --env ROS_MASTER_URI=http://172.17.0.1:11311 --env ROS_IP=172.17.0.2 \
 -v ~/.ros:/root  \
 introlab3it/rtabmap_ros:melodic-latest \
 roslaunch rtabmap_ros rtabmap.launch args:="-d --Icp/Force4DoF true" icp_odometry:=true rtabmapviz:=false database_path:=/root/rtabmap.db
```
is it crashing? My log:
```
... logging to /root/.ros/log/c4cda8d4-b372-11eb-87cc-9be2df5a7a4f/roslaunch-8a79cd3faeca-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.17.0.2:41633/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /rtabmap/icp_odometry/config_path: 
 * /rtabmap/icp_odometry/expected_update_rate: 0.0
 * /rtabmap/icp_odometry/frame_id: camera_link
 * /rtabmap/icp_odometry/ground_truth_base_frame_id: 
 * /rtabmap/icp_odometry/ground_truth_frame_id: 
 * /rtabmap/icp_odometry/guess_frame_id: 
 * /rtabmap/icp_odometry/guess_min_rotation: 0.0
 * /rtabmap/icp_odometry/guess_min_translation: 0.0
 * /rtabmap/icp_odometry/max_update_rate: 0.0
 * /rtabmap/icp_odometry/odom_frame_id: odom
 * /rtabmap/icp_odometry/publish_tf: True
 * /rtabmap/icp_odometry/queue_size: 10
 * /rtabmap/icp_odometry/scan_cloud_max_points: 0
 * /rtabmap/icp_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/icp_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: /root/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

[... 117 lines cut ...]
[ INFO] [1620859318.679541891]: rtabmap: Deleted database "/root/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1620859318.679595765]: rtabmap: Using database from "/root/rtabmap.db" (0 MB).
[ INFO] [1620859318.842326229]: rtabmap: Database version = "0.20.10".
[ INFO] [1620859318.842489077]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[ INFO] [1620859318.930993267]: /rtabmap/rtabmap: subscribe_depth = true
[ INFO] [1620859318.931110542]: /rtabmap/rtabmap: subscribe_rgb = true
[ INFO] [1620859318.931197782]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1620859318.931287740]: /rtabmap/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1620859318.931370312]: /rtabmap/rtabmap: subscribe_odom_info = true
[ INFO] [1620859318.931483496]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1620859318.931596696]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1620859318.931679801]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1620859318.931759441]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1620859318.931841857]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1620859318.931946034]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1620859318.932115736]: Setup depth callback
[ INFO] [1620859318.985211620]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
[ INFO] [1620859319.083062482]: rtabmap 0.20.10 started...
```

dagata-mining, 2021-05-12:
New Image seems to work, both with your script and my launch files