introlab/rtabmap #721: Miss mapping problem

jacklu333333, 2021-05-14:
Hi, I am using the rtabmap for my turtlebot. The robot model is kobuki and the camera model is intel realsense l515.

I use the configration on rtabmap ros page for l151. The following input is used for rtabmap.
/camera/imu
/camera/depth/image_rect_raw
/camera/color/image_raw
/camera/depth/color/points

Also, to have better accuracy of odometry, I convert the depth image into laserscan for kobuki base. 

However, this problem keeps happening when the robot starts to do rotation recovery to regain its own location. It will miss map the same wall multiple times and the more rotation the error accumulates. 
[image]

Is there any way I can solve it by adjusting the parameters?

matlabbe, 2021-05-15:
Are you using odometry from kuboki? Can you show the L515 point cloud and lidar at the same time in RVIZ? Does in general the costmap match correctly the laser scan?

jacklu333333, 2021-05-17:
For the odom I am sure. This is the screenshot
[image]

And this is the rviz with cost map. In the screenshot, I perform multiple times of rotation to induce the miss mapping 
[image]

In the beginning, everything maps perfectly. However, when the robot performs the rotation location recovery, the map did not link the info by space and feature, I have enabled both functions provided by rtamp but it seems it did not work as expected.

matlabbe, 2021-05-17:
Is this the local or global costmap? If it is the global costmap, can you share rtabmap database? (default `~/.ros/rtabmap.db`)

jacklu333333, 2021-05-18:
It is indeed the global costmap of the above picture. However, due to testing, I did not save the map above. Therefore, I reinduce the error. There might be a difference within the maps and the above screenshot.

There are two map files one is the default configuration. However, this will result in the following PCL error.
`Not enough correspondences found. Relax your threshold parameters.`

Hence, I fine-tune it with the configuration of the second file. 
The attachment file is my rtabmap launch file for my fine-tune.

This is the link
https://drive.google.com/drive/folders/1ksunjvzWvGPHi0u4o6btLgPw7usKi7TV?usp=sharing

Thank you for your help.

jacklu333333, 2021-05-26:
Hi, did miss config any parameter? 

Btw, I am also testing different configurations, and I notice that in the file I provided, scan_depth is not enabled. I'm sorry for that. 

However, in my experiment, it was not the reason for miss mapping. If the scan_depth is enabled, the problem remains.

matlabbe, 2021-05-30:
Hi,
Unfortunately the L515 doesn't work very well in this environment. See how much L515 is noisy (data inspected with rtabmap-databaseViewer):
[image]
[image]

With as much noise, the normal segmentation doesn't work very well (see the black pixels on left in the image above on the local grid map, light gray is empty). You could disable `Grid/NormalsSegmentation`. With other tuning (see the Parameters panel on right), we can filter much of the noise to avoid false obstacles on the floor:
[image]

Then for odometry, what do you use? The odometry seems drifting quite a lot, here between two consecutive frames the rotation was pretty bad:
[image]

jacklu333333, 2021-05-31:
Thanks for the information. I will dig into the noise problem.

As far as the odometer, I use the odometer provided by turtlebot package for kobuki base. I did suspect it was the problem. Hence I tried Reg/odom map-to-frame. However, it seems the drifting problem is so serious that it will not help much.

closed 2021-05-31 by jacklu333333