introlab/rtabmap #723: installing Rtabmap on Mac - segmentation fault upon running.

cryptoscientia, 2021-05-17:
Hello All, 

I have loaded Rtabmap onto my Mac using this link:
[https://github.com/introlab/rtabmap/wiki/Installation#source](url)

Everything seemed to run without much issue for the process.  When it finished, I ran the command, 'rtabmap' and get this error:

`% rtabmap
objc[15740]: Class vtkCocoaTimer is implemented in both /usr/local/opt/vtk/lib/libvtkRenderingUI-9.0.1.dylib (0x1299903d0) and /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e68eb18). One of the two will be used. Which one is undefined.
objc[15740]: Class vtkCocoaFullScreenWindow is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e68eb90) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x1298ddec8). One of the two will be used. Which one is undefined.
objc[15740]: Class vtkCocoaServer is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e68ebb8) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x1298ddef0). One of the two will be used. Which one is undefined.
objc[15740]: Class vtkCocoaGLView is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e68ec08) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x1298ddf40). One of the two will be used. Which one is undefined.
zsh: segmentation fault  rtabmap`

From what I gather there is two things happening here.  The first thing is some sort of duplicate file, of which I'm not sure what to do with it.  And the second is the segmentation fault.

I have looked it up several times and still have not come to much conclusion.  Please help me out.

matlabbe, 2021-05-17:
You have 2 vtks installed, which seems to produce conflicts. This could be the source of the seg fault.  With:
```
brew install pcl opencv@3
```
only VTK8 should be installed. Remove VTK9. For example, if you installed opencv (4), it depends on VTK9, which is why VTK9 may be also installed.

cryptoscientia, 2021-05-18:
Ran it:
` % brew install pcl opencv@3
Warning: pcl 1.11.1_8 is already installed and up-to-date.
To reinstall 1.11.1_8, run:
  brew reinstall pcl
Warning: opencv@3 3.4.14_1 is already installed and up-to-date.
To reinstall 3.4.14_1, run:
  brew reinstall opencv@3`

I also had uninstalled VTK9 and installed VTK8.  

I re-attempted the 'rtabmap' and it says:
`  % rtabmap
objc[20963]: Class vtkCocoaTimer is implemented in both /usr/local/opt/vtk/lib/libvtkRenderingUI-9.0.1.dylib (0x129be43d0) and /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e8aab18). One of the two will be used. Which one is undefined.
objc[20963]: Class vtkCocoaFullScreenWindow is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e8aab90) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x129b31ec8). One of the two will be used. Which one is undefined.
objc[20963]: Class vtkCocoaServer is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e8aabb8) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x129b31ef0). One of the two will be used. Which one is undefined.
objc[20963]: Class vtkCocoaGLView is implemented in both /usr/local/opt/vtk@8.2/lib/libvtkRenderingOpenGL2-8.2.1.dylib (0x11e8aac08) and /usr/local/opt/vtk/lib/libvtkRenderingOpenGL2-9.0.1.dylib (0x129b31f40). One of the two will be used. Which one is undefined.
zsh: segmentation fault  rtabmap`

Sorry it looks like that is not quite it ether.

matlabbe, 2021-05-18:
`/usr/local/opt/vtk/lib/libvtkRenderingUI-9.0.1.dylib` does it still really exist? If not, clear the cmake cache of rtabmap (delete all in build directory). If yes, your VTK9 is still installed for some reasons

cryptoscientia, 2021-05-18:
I cleared the cash and now it says:
`% rtabmap
dyld: Library not loaded: /usr/local/opt/opencv/lib/libopencv_gapi.4.5.dylib
  Referenced from: /usr/local/bin/rtabmap
  Reason: image not found
zsh: abort      rtabmap
`

matlabbe, 2021-05-19:
OpenCV 4.5 seems installed. cmake may be confused which one to choose to link on. Can you try without opencv 4.5 installed? (just opencv@3)