introlab/rtabmap #724: apk immediately crashes on LG G6 US997

AlexS2412, 2021-05-18:
im not sure why this happens because of th8s it makes it impossible to use the android side. 
[image]
[image]

matlabbe, 2021-05-19:
Wbich APK are you using?

AlexS2412, 2021-05-19:
> Wbich APK are you using?

the latest version in releases, RTABMap-0.20.8-tango.apk

matlabbe, 2021-05-19:
This is the Tango version, if your phone is not Tango, it will likely crash (telling that it didn't find tango library on the phone). For the android version, the unofficial one has to be installed from docker: https://github.com/introlab/rtabmap/wiki/Installation#docker- (follow those [steps ](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-tango-apk) but for android24 or android26 images). I know it is complicated, but the ARCore alone version is not very useful without a TOF camera. The android24 and android26 images exist mostly for testing (android24 can be used to compare Tango versus ARCore, android26 can be used to compare AREngine from Huawei versus ARCore) and to be ready when Google will decide to support officially TOF cameras in ARCore.