introlab/rtabmap #728: Increase texture mesh on mobile tango

Helmutfechners, 2021-05-28:
I want increase the resolution on Rtabmap app on my Tango mobile Asus Zenfone AR. 
When i scan objects I want that even text is readable.
Please check the quality of that photo. I was do a photo also with my mobile just with the normal camera app. I want perform that quality.
[image]

I use allready HD mode. I changed it the settings in Rtabmap app. Seems there was not enough pixels used!

Any sugesstions to get similar resolution like the camera can do?

Stay safe and health 
Best and thank you
Helmut

Helmutfechners, 2021-05-28:
Please check the screenshot of a scan sample with image from Rtabmap app:
[image]
here i can not read the printed text on the extra heavy paper folder

matlabbe, 2021-05-28:
Hi,
In HD Mode on Zenfone AR, the images recorded are 1920x1080 while the camera app take pictures at 5488x3088. Note that images may be downsampled when creating the meshes for visualization purpose, See Settings->Texture Resolution. The images in database are still saved in 1080p.

Here a comparison between 1080 (left taken by RTAB-Map Tango) vs 3088:
[image]
zooming:
[image]
zooming more:
[image]

Note also when exporting to optimized mesh, the output texture resolution setting is in Settings->Exporting->Texture size.

cheers,
Mathieu

Helmutfechners, 2021-05-28:
Hi Mathieu,
Thanks for your evaluation and quick response. One question more, is there an option to scan in lower distance than 1 meter to get a better resolution. I allready changed the scanning distance to 0 meter, but when scanning it show me notification that am to close to the object (nearer than 1 meter)
Happy to get an evaluation from you for that fact as well
Best Helmut

matlabbe, 2021-05-30:
Hi Helmut,

The 1 meter warning is only a warning, the data is still recorded, so you can scan closer than 1 meter. Note however that the depth sensor will eventually return nothing when too close from surfaces.