introlab/rtabmap #736: Mesh decimation (vtkQuadricDecimation) seems not working correctly with VTK9

matlabbe, 2021-06-19:
Here is an example with same export configs (limiting the mesh to 50 000 polygons). On top is the expected result, on bottom is the buggy result (some surfaces are missing and some vertexes are linked to a lot of polygons). The buggy version takes also a lot of time!

Working configurations:

- PCL 1.11.1 with VTK 8.2 (tested on Mac, homebrew)
- PCL 1.8.1 with VTK 6.2 (tested on Windows with appveyor [artifacts](https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts))

Not working configuration:

- Latest PCL (to this [date](https://github.com/PointCloudLibrary/pcl/commit/22cc0c6e47335cdb47356deab0490de1b411059e)) with VTK 9.0.1 (vcpkg)

[image]

We should test it under linux to make sure it is not a DLL/memory problem on Windows.

doublechenching, 2021-06-25:
I have the same problem when using vtk9.0. Numerical optimization problem? always print "unable factor linear system" last for several minutes

matlabbe, 2021-06-28:
The ubuntu version with VTK9 works as expected. The VTK6.2 built manually from source on Windows worked too. I checked VTK and PCL related code, and it seems to be the same for the past years. Maybe vcpkg version added some related [patches](https://github.com/microsoft/vcpkg/tree/master/ports/vtk) or have different build options than the default ones when built manually. Rebuilding VTK9 from source outside vcpkg could be a nice test to compare.