introlab/rtabmap #738: problems connecting (apps) RTAB-Map with LiDAR

hyemhyemu, 2021-06-23:
Hello, 

I am trying to connect RTAB-Map - 3D LiDAR Scanner (installed in iPhone 8) with Intel REALSENSE LiDAR Camera L515.
I tried to connect iPhone 8 and L515 with a lighting cable - type C, but the sense's image didn't show up on the apps. 
I am aiming to do a room 3D scan with RTAB-Map via iPhone, please help!

Thanks you all.

closed 2021-06-25 by hyemhyemu

CARRTUSF, 2021-07-14:
Hi, I am trying to do the same. However I am trying to do it over an iPad. Can we get in touch?

hyemhyemu, 2021-07-15:
> Hi, I am trying to do the same. However I am trying to do it over an iPad. Can we get in touch?

I looked up the YouTube video about L515 and I don't think it can neither link up nor work with any iPhone/iPad device, even with the installed RTAB-MAP apps. The RTAB-MAP apps will work with the LiDAR function on iPhone 12 itself, instead of an external Intel device.

matlabbe, 2021-07-30:
Yes the current iOS app only works with internal LiDAR from iPhone 12 Pro and iPad Pro.