introlab/rtabmap #739: Docker build android aar instructions

TheBricktop, 2021-06-23:
Hello,
i have recently started process of building RTABmap as a plugin library for unity into dll / so / aar format.
I have some problems at start because ive wanted to build android from docker/bionic for testing and  hit the wall 

> => ERROR [ 8/10] ADD deps.bash /root/deps.bash                                                                    0.0s
>------
 > [ 8/10] ADD deps.bash /root/deps.bash:
>------
>failed to compute cache key: "/deps.bash" not found: not found

the deps.bash is present in the android folder but it still cant find it

matlabbe, 2021-07-15:
If you move to android folder before doing docker build?
```
cd android
docker build -t rtabmap:android .
```

TheBricktop, 2021-08-13:
Docker file has build correctly, but im not exactly sure how to proceed, since my build needs realsense library (i have one prebuild for android and win). 
Do i have to create my own docker image for rtabmap android without Tango and with realsense enabled?
Im pretty stumped.

matlabbe, 2021-08-13:
Android app doesn't support realsense.

TheBricktop, 2021-08-14:
Could it be possible to build Android APK or aar of rtabmap without tango and with realsense enabled?

matlabbe, 2021-09-03:
The android APK can be built without Tango, there is an experimental build ([android26](https://github.com/introlab/rtabmap/wiki/Installation#docker) docker image) working on Samsung Note10+ (using ARCore + TOF camera). 

For your question yes, but some coding is required on the app side to select realsense2 driver. Theoretically, we could combine ARCore pose estimation with L515 LiDAR data for example, or even use only RealSense D435i or D455 independently of ARCore (however visual odometry perfomance on phones without ARCore may not be super fast and may result in losing tracking easily).

I've never tried to build realsense2 SDK for android NDK, but if [CameraRealSense2](https://github.com/introlab/rtabmap/blob/master/corelib/src/camera/CameraRealSense2.cpp) class can be re-used as is, it would save a lot of integration time.

TheBricktop, 2021-09-06:
librealsense is pretty easy to build considering that im just dabbling in NDK and still managed to integrate it into android app.
I have some build artifacts if You want to check it fast.

matlabbe, 2021-09-08:
Yes it is possible, contact me on my email (same username as here @ gmail).

TheBricktop, 2021-09-13:
Hello, @matlabbe  i've sent You an email about it.

Gvascons, 2022-10-28:
Hello,
I'm currently trying to build an Rtabmap Android APK that supports Intel Real Sense cameras - hence librealsense (in order to leverage it's LiDAR depth technology capabilities). This issue seems to be associated with your discussion. Could you provide any updates (if any) on this Rtabmap-RealSense integration?
Best regards

matlabbe, 2022-11-27:
No updates from my part for android realsense integration.