introlab/rtabmap #740: symbol lookup error: librtabmap_core.so.0.20 (Jetson)

guglielmodev, 2021-06-23:
When I run `rtabmap` on a Jetson, it crashes with the following error:

    rtabmap: symbol lookup error: /usr/local/lib/librtabmap_core.so.0.20: undefined symbol:
    _ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt10shared_ptrINS_32AbstractHyperGraphElementCreatorEE

built from source (v0.20.9) on the board without errors

matlabbe, 2021-07-15:
are there multiple g2o libraries installed? On runtime, it may be loading the wrong g2o library (not the one built against), so there is a missing symbol.

guglielmodev, 2021-09-27:
Sorry for the late reply, didn't saw your message, I'll check the g2o installation