introlab/rtabmap #744: G2O graph optimization cannot be used with g2o built from ORB_SLAM, only SBA is available.

JSaunders97, 2021-07-01:
Hello,

I'm trying to build rtabmap with ORB SLAM3, I have built the code successfully, however when I run it I get the following error:

[ERROR] (2021-07-01 15:04:55.323) OptimizerG2O.cpp:1169::optimize() G2O graph optimization cannot be used with g2o built from ORB_SLAM, only SBA is available.
[ WARN] (2021-07-01 15:04:55.323) Rtabmap.cpp:4454::optimizeGraph() Optimization has failed (poses=15, guess=2, links=14)...
[ WARN] (2021-07-01 15:04:55.323) Rtabmap.cpp:4375::optimizeCurrentMap() Failed to optimize the graph! returning empty optimized poses...
[ WARN] (2021-07-01 15:04:55.323) Rtabmap.cpp:3110::process() Graph optimization failed! Rejecting last loop closures added.
[ WARN] (2021-07-01 15:04:55.323) Rtabmap.cpp:3114::process() Loop closure 830->756 rejected!

I've tried rebuilding with cvsba but that doesn't seem to have solved the problem. What's the problem and how do I solve it?

Many thanks!

JSaunders97, 2021-07-01:
SOLVED!

When launching rtabmap_ros using rtabmap.launch it loads the default database because of this line:

<arg name="database_path"           default="~/.ros/rtabmap.db"/>

When this is done, some parameters are overridden including Optimizer/Strategy. In my case this was set to g2o, hence the aforementioned error. To fix this, change the line above to: 

<arg name="database_path"           default=""/>

This will prevent the default database overriding parameters.

closed 2021-07-01 by JSaunders97

matlabbe, 2021-07-01:
Without a database path, the database is created in memory and not saved on exit. If it is what you want it is fine. Otherwise, you can do `roslaunch rtabmap_ros rtabmap.launch args:="-d" ...` to overwrite the previous database (it will delete the old one and make a new one). After exit, it is then possible to debug `~/.ros/rtabmap.db` if something went wrong.

JSaunders97, 2021-07-01:
Thank you for the advice, I will want to save the database in the future but I'm just testing at the moment.