introlab/rtabmap #747: On the use of two D435 + rtabmap packages, and rtabmap not compile

ju-mingyue, 2021-07-05:
Hello, I have a problem when using two D435 cameras in combination with the rtabmap program. My program configuration is as follows:
`
 
     <launch>
    
    
    
    <arg name="strategy"        default="0" />
    <arg name="feature"         default="6" />
    <arg name="nn"              default="3" />
    <arg name="max_depth"       default="4.0" />
    <arg name="min_inliers"     default="20" />
    <arg name="inlier_distance" default="0.02" />
    <arg name="local_map"       default="1000" />
    <arg name="odom_info_data"  default="true" />
    <arg name="wait_for_transform"  default="true" />
    
    <group ns="rtabmap">
      
      <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="">
        <param name="subscribe_depth"  type="bool"   value="false"/>
        <param name="subscribe_rgbd"   type="bool"   value="true"/>
        <param name="rgbd_cameras"    type="int"    value="2"/>
        <param name="frame_id"         type="string" value="base_link"/>
        <param name="odom_frame_id"       type="string" value="/cart_odom_frame"/>
        <param name="gen_scan"         type="bool"   value="false"/>
        <param name="wait_for_transform" type="bool"   value="$(arg wait_for_transform)"/>
        <param name="map_negative_poses_ignored" type="bool"   value="false"/>        
        <param name="map_negative_scan_empty_ray_tracing" type="bool" value="false"/> 
        <remap from="rgbd/image0"       to="/d435_L/color/image_raw"/>
        <remap from="rgbd/image1"       to="/d435_R/color/image_raw"/>
        <remap from="depth/image0"       to="/d435_L/depth/image_raw"/>
        <remap from="depth/image1"       to="/d435_R/depth/image_raw"/>
        <remap from="rgbd/camera_info0"   to="/d435_L/color/camera_info"/>
        <remap from="rgbd/camera_info1"   to="/d435_R/color/camera_info"/>
        <param name="Grid/FromDepth"     type="string" value="false"/>
        <param name="Vis/EstimationType" type="string" value="0"/> 
        <param name="Vis/MinInliers"     type="string" value="10"/>
        <param name="Vis/InlierDistance" type="string" value="$(arg inlier_distance)"/>
      </node>
    </group>
      </launch>
`

Problems after running as follows:
[image]

[image]

I have looked for a lot of solutions, but still can't solve it, whether it is downloading the source package or using apt-get to download.

I have tried this link,But to no avail
[https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i](url)

taletovic, 2021-07-19:
Hi @ju-mingyue, in order to use that funktion RTAB-Map has to be built from source like this:
```
$ sudo apt remove ros-$ROS_DISTRO-rtabmap-ros
$ cd ~/catkin_ws/src
$ git clone -b $ROS_DISTRO-devel https://github.com/introlab/rtabmap_ros.git
$ cd ~/catkin_ws
$ catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON
```

ju-mingyue, 2021-07-19:
@taletovic 
Thank you very much for your help. Now I have a new problem. My D435 camera will suddenly detect a non-existent obstacle and add it to the costmap, causing the robot to change its path.

matlabbe, 2021-07-20:
Following parameters could be used to filter noise:
```
rtabmap --params | grep Grid/
Param: Grid/NoiseFilteringMinNeighbors = "5"               [Noise filtering minimum neighbors.]
Param: Grid/NoiseFilteringRadius = "0.0"                   [Noise filtering radius (0=disabled). Done after segmentation.]
```

ju-mingyue, 2021-07-21:
@matlabbe 
I'm very sorry, I gave up using rtabmap. I chose the simplest method to convert the depth information of the d435 camera into a laser signal. In the end, the above problem appeared. With regard to the current method, do you have a way to solve the noise problem?

matlabbe, 2021-08-18:
If you generate a 2d scan, you could easily filter the scan data by comparing previous and next ranges to find ranges that are noise and set them to 0.

hariharan382, 2022-07-19:
Please if you solved the issue please let me know,I don't know how to adjust the time stamp of the rtabmap inputs