introlab/rtabmap #749: Export RGB,Depth and camera info

Zumbalamambo, 2021-07-06:
Thanks for the clean app for iOS. Is it possible to export rgb,depth images, and the camera extrinsic , intrinsic matrix from the app?

matlabbe, 2021-07-07:
From the ios app no, but you can do it with RTAB-Map Desktop though (with rtabmap-databaseViewer or rtabmap-export tool), by loading the database created on ios.

To export images with ids and poses of the camera (pose format 11 is the most flexible, it includes timestamps and ids, while being also compatible with [RGB-D dataset evaluation](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools) tool):
```
rtabmap-export --images_id --poses_camera --poses_format 11 rtabmap.db
```
To export images with timestamps and poses of the camera:
```
rtabmap-export --images --poses_camera --poses_format 11 rtabmap.db
```

From rtabmap-databaseViewer, you have File->Extract images... and File->Extract poses... options:
[image]

Note that I updated recently the export options to export calibration for each individual image (was exporting only one calibration file before, assuming calibration of the camera didn't change for the map). If you need calibration for each image, update to latest code.

Zumbalamambo, 2021-07-09:
@matlabbe  thank you!!!... Do I have to do any "loop closure detection and other such steps in the desktop as well? I have done it before transferring the database to my desktop so I presume that the database is overridden with the optimized poses.

matlabbe, 2021-07-28:
Hi,
1. The post-processing stuff done in the App are saved in the database, so yes when opening in Desktop there should be the same number of constraints.
2. It is the reference frame. For iOS databases, "base" will export the poses in ros coordinate frame (x->forward, y->left and z->up), while "camera" will export the pose of the camera in optical frame (x->right, y->down, z->forward).
3. With the UI option above, you will be asked in PNG or JPG. For command line tool: RGB images are exported in JPG (they are already saved in JPG in the database, so exporting in PNG won't change anything unless `Mem/ImageCompressionFormat` parameter was set to `.png`) and depth images are in PNG.

closed 2021-11-22 by Zumbalamambo

matlabbe, 2022-05-09:
You should get the*.yaml files for camera calibration corresponding to each image when extracting the images. The projection matrix is included.

JetBelson, 2022-05-31:
> You should get the*.yaml files for camera calibration corresponding to each image when extracting the images. The projection matrix is included.

I tried to export the calibration information from the 'Data' table in db file directly. 
After decoding the bytes, the projection matrix is missing in the field of "calibration".
I got the below header information.
(0, 20, 19, 0, 720, 960, 9, 0 ,0 , 0 , 12)

Referring to the CameraModel::serialize() function, the size of the projection matrix is zero. 
I think I have missed some critical points. Can you advise? Thanks!

matlabbe, 2022-09-06:
Matrix K is there. With K, you can project pixels in camera optical frame to create a point cloud.