introlab/rtabmap #753: [FATAL] (2021-07-09 16:58:42.406) Memory.cpp:4143::createSignature() && [rtabmap/rtabmap-1] process has died [pid 26951, exit code -6, cmd

Haydar1990, 2021-07-09:
Hallo , i am new on ROS & Rtabmap, i am trying to bild a 3D map using Gazebo and Rtabmap And i got this 

My Launch file:
<launch>
  <arg name="database_path" default="~/Documents/RTAB-Map/rtabmap.db"/>
  <arg name="localization" default="false"/>
  <arg name="args" default=""/>

  <arg name="wait_for_transform" default="0.2"/>
  
  
  

  
  <group ns="rtabmap">
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
          <param name="database_path"  type="string" value="$(arg database_path)"/>
          <param name="frame_id"       type="string" value="base_footprint"/>
          <param name="odom_frame_id"       type="string" value="odom_comb"/>
          <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
          <param name="subscribe_depth"     type="bool"   value="true"/>
          <param name="subscribe_scan"      type="bool"   value="true"/>

          
          <remap from="scan"            to="/scan"/>
          <remap from="rgb/image"       to="/camera/color/image_raw"/>
          <remap from="depth/image"     to="/camera/depth/image_raw"/>
          <remap from="rgb/camera_info" to="/camera/color/camera_info"/>

          
          <remap from="grid_map" to="/map"/>

          
          <param name="RGBD/ProximityBySpace"        type="string" value="true"/>   
          <param name="RGBD/OptimizeFromGraphEnd"    type="string" value="false"/>  
          <param name="Kp/MaxDepth"                  type="string" value="4.0"/>
          <param name="Reg/Strategy"                 type="string" value="1"/>      
          <param name="Icp/CorrespondenceRatio"      type="string" value="0.3"/>
          <param name="Vis/MinInliers"               type="string" value="5"/>      
          <param name="Vis/InlierDistance"           type="string" value="0.1"/>    
          <param name="RGBD/AngularUpdate"           type="string" value="0.01"/>    
          <param name="RGBD/LinearUpdate"            type="string" value="0.01"/>    
          <param name="Rtabmap/TimeThr"              type="string" value="700"/>
          <param name="Mem/RehearsalSimilarity"      type="string" value="0.30"/>
          <param name="Optimizer/Slam2D"             type="string" value="true"/>
          <param name="Reg/Force3DoF"                type="string" value="true"/>
          
          <param if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
          <param if="$(arg localization)" name="Mem/InitWMWithAllNodes" type="string" value="true"/>
    </node>
  </group>
</launch>

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /rtabmap/rtabmap/Icp/CorrespondenceRatio: 0.3
 * /rtabmap/rtabmap/Kp/MaxDepth: 4.0
 * /rtabmap/rtabmap/Mem/RehearsalSimilarity: 0.30
 * /rtabmap/rtabmap/Optimizer/Slam2D: true
 * /rtabmap/rtabmap/RGBD/AngularUpdate: 0.01
 * /rtabmap/rtabmap/RGBD/LinearUpdate: 0.01
 * /rtabmap/rtabmap/RGBD/OptimizeFromGraphEnd: false
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 1
 * /rtabmap/rtabmap/Rtabmap/TimeThr: 700
[... 226 lines cut ...]
/rtabmap/local_grid_empty
/rtabmap/local_grid_ground
/rtabmap/local_grid_obstacle
/rtabmap/local_path
/rtabmap/local_path_nodes
/rtabmap/localization_pose
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapPath
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_global_frontier_space
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/proj_map
/rtabmap/scan_map
/rtabmap/user_data_async
/scan
/tf
/tf_static

Thank You in advance

matlabbe, 2021-07-09:
Hi, rtabmap is expecting a RGB image always equal or larger than the depth image. In your case, the depth image is 2x larger than the RGB image. Depth= 1280x720 and RGB=640x480. Do you have a RGB topic at 1280x720 resolution? or a depth topic at 640x480 resolution that you could use instead?

Haydar1990, 2021-07-11:
Thank you very much, I already have the file d435.xarco edited and it works.

closed 2021-07-11 by Haydar1990