introlab/rtabmap #759: issue when installing rtabmap_ros

Skander1997, 2021-08-04:
With ros noetic Ubuntu 20.04 LTS I have an error: 
[ 64%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
/home/skander/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp: In constructor ‘MapsManager::MapsManager()’:
/home/skander/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:72:16: error: invalid use of incomplete type ‘class rtabmap::OctoMap’
   72 |   octomap_(new OctoMap),
      |                ^~~~~~~
In file included from /home/skander/catkin_ws/src/rtabmap_ros/src/MapsManager.cpp:27:
/home/skander/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/MapsManager.h:40:7: note: forward declaration of ‘class rtabmap::OctoMap’
   40 | class OctoMap;
      |       ^~~~~~~
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:76: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3524: rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Invoking "make -j1 -l1" failed

how to fix the class OctoMap ? 
I downloaded Rtabmap and Rtabmap from source.
Thank You

Majed-Alsubaie, 2021-08-04:
Hi,
Make sure you downloaded Rtabmap_ROS and Rtabmap for your ROS  version  
just clone 
1-`git clone https://github.com/introlab/rtabmap --branch noetic-devel `  //rtabmap noetic
2- `git clone https://github.com/introlab/rtabmap_ros --branch noetic-devel ` //rtab,ap_ros noetic

Skander1997, 2021-08-04:
Project 'open_chisel' specifies
  '/home/skander/catkin_ws/src/OpenChisel/open_chisel/include' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in
  '/home/skander/catkin_ws/src/OpenChisel/open_chisel//home/skander/catkin_ws/src/OpenChisel/open_chisel/include'.
  Report the problem to the maintainer ' <mklingen@andrew.cmu.edu>' and
  request to fix the problem.

### sorry I had with open_chisel, i tried to install it but it didn't work so i deleted and still have the same problem, this problem is when i try cmake ..
thank you for your help

Skander1997, 2021-08-04:
@Majed-Alsubaie

matlabbe, 2021-09-02:
Make sure to uninstall rtabmap binaries:
``` 
sudo apt remove ros-noetic-rtabmap
```
to avoid rtabmap_ros trying to link on rtabmap binaries instead of the one built from source.

Then make sure you are using same branch name between rtabmap and rtabmap_ros projects.

cedricgoubard, 2021-10-21:
Hi, 

I am having the same issue; here is the output from `catkin build`:

```bash
Failed <<< rtabmap_ros                                  [ 2 minutes and 52.8 seconds ].. 
#31 728.2 [build] Summary: 78 of 79 packages succeeded.                                  
#31 728.2 [build] Ignored: None.                                                         
#31 728.2 [build] Warnings: 2 packages succeeded with warnings.                          
#31 728.2 [build] Abandoned: No packages were abandoned.                                 
#31 728.2 [build] Failed: 1 packages failed.                                             
#31 728.2 [build] Runtime: 12 minutes and 5.8 seconds total.                             
#31 728.2 [build] Note: Workspace packages have changed, please re-source setup files to use them.
#31 728.2 _______________________________________________________________________________
#31 728.2 Warnings << pcl_conversions:cmake /root/ros_ws/logs/pcl_conversions/build.cmake.000.log
#31 728.2 ** WARNING ** io features related to ensenso will be disabled
#31 728.2 ** WARNING ** io features related to davidSDK will be disabled
#31 728.2 ** WARNING ** io features related to dssdk will be disabled
#31 728.2 ** WARNING ** io features related to pcap will be disabled
#31 728.2 ** WARNING ** io features related to png will be disabled
#31 728.2 ** WARNING ** io features related to libusb-1.0 will be disabled
#31 728.2 cd /root/ros_ws/build/pcl_conversions; catkin build --get-env pcl_conversions | catkin env -si  /usr/bin/cmake /root/ros_ws/src/perception_pcl/pcl_conversions --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/ros_ws/devel/.private/pcl_conversions -DCMAKE_INSTALL_PREFIX=/root/ros_ws/install; cd -
#31 728.2 
#31 728.2 ...............................................................................
#31 728.2 _______________________________________________________________________________
#31 728.2 Warnings << pcl_ros:cmake /root/ros_ws/logs/pcl_ros/build.cmake.000.log        
#31 728.2 ** WARNING ** io features related to ensenso will be disabled
#31 728.2 ** WARNING ** io features related to davidSDK will be disabled
#31 728.2 ** WARNING ** io features related to dssdk will be disabled
#31 728.2 ** WARNING ** io features related to pcap will be disabled
#31 728.2 ** WARNING ** io features related to png will be disabled
#31 728.2 ** WARNING ** io features related to libusb-1.0 will be disabled
#31 728.2 cd /root/ros_ws/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/bin/cmake /root/ros_ws/src/perception_pcl/pcl_ros --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/ros_ws/devel/.private/pcl_ros -DCMAKE_INSTALL_PREFIX=/root/ros_ws/install; cd -
#31 728.2 
#31 728.2 ...............................................................................
#31 728.2 _______________________________________________________________________________
#31 728.2 Errors << rtabmap_ros:make /root/ros_ws/logs/rtabmap_ros/build.make.000.log    
#31 728.2 /root/ros_ws/src/rtabmap_ros/src/MapsManager.cpp: In constructor 'MapsManager::MapsManager()':
#31 728.2 /root/ros_ws/src/rtabmap_ros/src/MapsManager.cpp:72:16: error: invalid use of incomplete type 'class rtabmap::OctoMap'
#31 728.2    octomap_(new OctoMap),
#31 728.2                 ^~~~~~~
#31 728.2 In file included from /root/ros_ws/src/rtabmap_ros/src/MapsManager.cpp:27:0:
#31 728.2 /root/ros_ws/src/rtabmap_ros/include/rtabmap_ros/MapsManager.h:40:7: note: forward declaration of 'class rtabmap::OctoMap'
#31 728.2  class OctoMap;
#31 728.2        ^~~~~~~
#31 728.2 make[2]: *** [CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
#31 728.2 make[2]: *** Waiting for unfinished jobs....
#31 728.2 make[1]: *** [CMakeFiles/rtabmap_ros.dir/all] Error 2
#31 728.2 make: *** [all] Error 2
#31 728.2 cd /root/ros_ws/build/rtabmap_ros; catkin build --get-env rtabmap_ros | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -
#31 728.2 
#31 728.2 ...............................................................................
#31 728.2 Failed << rtabmap_ros:make                              [ Exited with code 2 ] 
------
executor failed running [/bin/bash -c source /opt/ros/noetic/setup.bash && catkin build]: exit code: 1
```

I installed both `rtabmap` and `ratbmap_ros` from source:
```Dockerfile
# Extract from my Dockerfile
RUN git clone https://github.com/introlab/rtabmap.git --branch noetic-devel && \
    cd rtabmap/build && \
    cmake .. && \
    make -j4 && \
    make install

### ... ###

RUN git clone https://github.com/ros-perception/perception_pcl.git &&\
        cd perception_pcl && git checkout melodic-devel && cd .. && \
    git clone https://github.com/introlab/rtabmap_ros.git  --branch noetic-devel

RUN source /opt/ros/noetic/setup.bash && catkin build 
```

Finally, `ros-noetic-rtabmap` is not installed:
```bash
root@wheelchair:/navigation/rtabmap# apt list --installed | grep rtab
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

root@wheelchair:/navigation/rtabmap# 
```

Do you have any suggestions?

Thank you for your help.

matlabbe, 2021-10-21:
It is because rtabmap was not built with OctoMap dependency. You can install octomap with:
```
sudo apt install ros-noetic-octomap
```
However, it should have just ignored it with a compilation error. I fixed the error in the commit above.