introlab/rtabmap #760: Program crash upon exiting visualizer

parkerlreed, 2021-08-05:
Regarding https://github.com/introlab/rtabmap/issues/425#issuecomment-516085692

I finally got around to testing this on an Ubuntu setup. 

Rundown: Performing a scan (Freenect2 in my case) and running a visualization works but upon exiting the visualizer window the entire program exits. 

Ubuntu 20.04
ROS Noetic
rtabmap-ros (still using as standalone/had to recompile in ros mode for freenect2)
All of the deps from the noetic package

I built in debug mode and installed all the dbgsym packages I could think of, but it doesn't seem to segfault so I have been unable to get a backtrace,

Judging by the current output, it may have never been a vtk issue to begin with.

```
parker@parusb:~$ rtabmap
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Program started...
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 23 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:6 with serial 067713440247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ WARN] (2021-08-04 19:55:29.401) CameraModel.cpp:358::load() Could not load calibration file "/home/parker/Documents/RTAB-Map/camera_info/067713440247_depth.yaml".
[ WARN] (2021-08-04 19:55:29.401) CameraModel.cpp:358::load() Could not load calibration file "/home/parker/Documents/RTAB-Map/camera_info/067713440247_rgb.yaml".
[ WARN] (2021-08-04 19:55:29.401) CameraFreenect2.cpp:258::init() Missing calibration files for camera "067713440247" in "/home/parker/Documents/RTAB-Map/camera_info" folder, default calibration is used. Note that from version 0.11.10, calibration suffixes for Freenect2 driver have changed from "_left"->"_depth" and "_right"->"_rgb". You can safely rename the calibration files to avoid recalibrating.
[Info] [OpenGLDepthPacketProcessor] avg. time: 12.4649ms -> ~80.2254Hz
[ WARN] (2021-08-04 19:55:33.405) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 6/20 (matches=72) between -1 and 3" (guess=xyz=-0.002641,0.003865,0.005283 rpy=-0.004711,0.003083,-0.006551)
[ WARN] (2021-08-04 19:55:33.405) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.002641,0.003865,0.005283 rpy=-0.004711,0.003083,-0.006551), trying again without a guess.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.0976ms -> ~39.8444Hz
[ WARN] (2021-08-04 19:55:34.300) OdometryF2M.cpp:539::computeTransform() Trial with no guess succeeded!
[ WARN] (2021-08-04 19:55:34.612) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 11/20 (matches=157) between -1 and 4" (guess=xyz=-0.009523,0.000790,0.001782 rpy=-0.010381,-0.001233,-0.060784)
[ WARN] (2021-08-04 19:55:34.612) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.009523,0.000790,0.001782 rpy=-0.010381,-0.001233,-0.060784), trying again without a guess.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 12.4006ms -> ~80.6411Hz
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[ WARN] (2021-08-04 19:55:37.154) OdometryF2M.cpp:539::computeTransform() Trial with no guess succeeded!
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x300017d
  Serial number of failed request:  5217
  Current serial number in output stream:  5218
```

parkerlreed, 2021-08-05:
Judging by the odd behavior of no segfault, could the visualizer window somehow be inheriting the window id of the main window and exiting both as a result?

matlabbe, 2021-09-02:
For "running a visualization works but upon exiting the visualizer window the entire program exits.", do you mean the visualization shown when doing Edit->View point clouds?

Currently I have an issue on Ubuntu 20.04 default VTK version, which crashes rtabmap everytime we close a visualization window in which we enabled EDL shader. The crash doesn't happen with more recent VTK versions.

parkerlreed, 2021-09-03:
Yeah exactly that. View point clouds is the path that leads to the crash condition when closing the window. I can at least confirm it still does it on VTK 8.2.0 (if that is the issue)

matlabbe, 2021-09-05:
I cannot reproduce the problem with:
```
$ rtabmap --version
RTAB-Map:              0.20.13
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
```
when opening a database and do Edit->View Clouds. Just if I enable EDL shader in the view cloud window then close it:
```
ERROR: In /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkFrameBufferObject.cxx, line 222
vtkFrameBufferObject (0x5641957fff40): Frame buffer object was not initialized correctly.

ERROR: In /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkFrameBufferObject.cxx, line 1029
vtkFrameBufferObject (0x5641957fff40): The framebuffer is incomplete : FRAMEBUFFER_INCOMPLETE_ATTACHMENT

Current framebuffer is bind to framebuffer object 1
color attachement 0:
 this attachment is a texture with name: 1
 its mipmap level is: 0
 this is not a cube map texture.
 this is not 3D texture.
color attachement 1:
 this attachment is empty
color attachement 2:
 this attachment is empty
color attachement 3:
 this attachment is empty
color attachement 4:
 this attachment is empty
color attachement 5:
 this attachment is empty
color attachement 6:
 this attachment is empty
color attachement 7:
 this attachment is empty
depth attachement :
 this attachment is a texture with name: 2
 its mipmap level is: 0
 this is not a cube map texture.
 this is not 3D texture.
stencil attachement :
 this attachment is empty
there are 8 draw buffers. 
draw buffer[0]=GL_COLOR_ATTACHMENT0
draw buffer[1]=GL_NONE
draw buffer[2]=GL_NONE
draw buffer[3]=GL_NONE
draw buffer[4]=GL_NONE
draw buffer[5]=GL_NONE
draw buffer[6]=GL_NONE
draw buffer[7]=GL_NONE
read buffer=GL_COLOR_ATTACHMENT0
ERROR: In /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkOpenGLRenderer.cxx, line 538
vtkOpenGLRenderer (0x564194b406a0): failed after Clear 1 OpenGL errors detected
  0 : (1286) Invalid framebuffer operation

ERROR: In /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkFrameBufferObject.cxx, line 222
vtkFrameBufferObject (0x5641a2817d30): Frame buffer object was not initialized correctly.

ERROR: In /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkFrameBufferObject.cxx, line 1029
vtkFrameBufferObject (0x5641a2817d30): The framebuffer is incomplete : FRAMEBUFFER_INCOMPLETE_ATTACHMENT

Current framebuffer is bind to framebuffer object 2
color attachement 0:
 this attachment is a texture with name: 3
 its mipmap level is: 0
 this is not a cube map texture.
 this is not 3D texture.
color attachement 1:
 this attachment is empty
color attachement 2:
 this attachment is empty
color attachement 3:
 this attachment is empty
color attachement 4:
 this attachment is empty
color attachement 5:
 this attachment is empty
color attachement 6:
 this attachment is empty
color attachement 7:
 this attachment is empty
depth attachement :
 this attachment is empty
stencil attachement :
 this attachment is empty
there are 8 draw buffers. 
draw buffer[0]=GL_COLOR_ATTACHMENT0
draw buffer[1]=GL_NONE
draw buffer[2]=GL_NONE
draw buffer[3]=GL_NONE
draw buffer[4]=GL_NONE
draw buffer[5]=GL_NONE
draw buffer[6]=GL_NONE
draw buffer[7]=GL_NONE
read buffer=GL_COLOR_ATTACHMENT0
rtabmap: /build/vtk7-yd0MKW/vtk7-7.1.1+dfsg2/Rendering/OpenGL2/vtkFrameBufferObject.cxx:908: void vtkFrameBufferObject::RenderQuad(int, int, int, int, vtkShaderProgram*, vtkOpenGLVertexArrayObject*): Assertion `"pre increasing_x" && minX<=maxX' failed.
Aborted (core dumped)
```

With the following build, there is no crash:
```
./rtabmap --version
RTAB-Map:              0.20.13
PCL:                1.11.1-dev
With VTK:                9.0.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With Madgwick:            true
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:               true
With OpenNI2:             true
With Freenect:            true
With Freenect2:           true
With K4W2:               false
With K4A:                 true
With DC1394:              true
With FlyCapture2:        false
With ZED:                 true
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:             true
With libpointmatcher:     true
With CCCoreLib:           true
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

parkerlreed, 2021-09-05:
So in relation to #769, PCL 1.10 is what I need? I've been unable to retest the crash as I cannot compile rtabmap against PCL 1.11/1.12/master (It was working on 1.11 at some point)

VTK may be harder as 8.2.0 is what my system ships with. The struggle of up-to-date libraries continues :/

matlabbe, 2021-09-06:
At some point, PCL maintainers moved from VTK6-7-8 to VTK9 at PCL 1.11. It is unclear which PCL version is the most compatible with VTK 8.2, as default VTK version on Ubuntu 18.04 is VTK6, then on Ubuntu 20.04 it is VTK7.

parkerlreed, 2021-09-06:
Video 

https://streamable.com/rrc7zr

```
[parker@wolfcola ~]$ rtabmap --version
RTAB-Map:              0.20.13
PCL:                    1.12.0
With VTK:                8.2.0
OpenCV:                  4.5.3
With OpenCV xfeatures2d:  true
With OpenCV nonfree:      true
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With Madgwick:            true
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:           true
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

matlabbe, 2021-09-08:
Do you know which Qt version it is used on that machine? I'll start a build with VTK 8.2 to see...

parkerlreed, 2021-09-08:
QMake version 3.1
Using Qt version 5.15.2 in /usr/lib

matlabbe, 2021-09-08:
Did a test with VTK 8.2.0:
```
./rtabmap --version
RTAB-Map:              0.20.13
PCL:                1.12.0-dev
With VTK:                8.2.0
OpenCV:                  4.2.0
```
no problem. I did the same steps than in your video, no crash. The visualization window did appear in front of the main window. On Ubuntu 20.04, it is Qt 5.12.8. On my Mac build, I do have the problem of the visualization window appearing behind the main window, Qt version is 5.15.2, however at that time I didn't push more the debugging because it is on Mac Silicon, so I was expecting that Qt would be not stable yet, but looks like it is more a Qt 5.15.2 compatibility problem (independent of the os). But there is no crash at least, here is the config on the Mac ([last time I built with default homebrew packages](https://github.com/introlab/rtabmap/wiki/Installation#source)):
 ```
./rtabmap --version
RTAB-Map:              0.20.13
PCL:                    1.11.1
With VTK:                8.2.0
OpenCV:                 3.4.14
```

parkerlreed, 2021-09-10:
Strange. Thanks for checking. Still trying to see if I can get a good breakpoint debug trace. It doesn't segfault so it's been trickier trying to hook the correct spot.

parkerlreed, 2021-09-10:
So on a smaller scan it doesn't crash fully but locks up when exiting visualizer.

Seems to get stuck in xcb_wait_for_special_event. I can continue to Ctrl C out of the loop and it just repeats.

```
Single stepping until exit from function xcb_wait_for_special_event,
which has no line number information.
[Thread 0x7fffbeffd640 (LWP 9169) exited]
[Thread 0x7fffbf7fe640 (LWP 9168) exited]
[Thread 0x7fffbffff640 (LWP 9167) exited]
[Thread 0x7fffc4be7640 (LWP 9166) exited]
^C
Thread 1 "rtabmap" received signal SIGINT, Interrupt.
0x00007ffff58b4b2f in poll () from /usr/lib/libc.so.6
(gdb) step
Single stepping until exit from function poll,
which has no line number information.
0x00007fffde45845b in xcb_wait_for_special_event () from /usr/lib/libxcb.so.1
(gdb) step
Single stepping until exit from function xcb_wait_for_special_event,
which has no line number information.
^C
Thread 1 "rtabmap" received signal SIGINT, Interrupt.
0x00007ffff58b4b2f in poll () from /usr/lib/libc.so.6
(gdb) step
Single stepping until exit from function poll,
which has no line number information.
0x00007fffde45845b in xcb_wait_for_special_event () from /usr/lib/libxcb.so.1
(gdb) step
Single stepping until exit from function xcb_wait_for_special_event,
which has no line number information.
^C
```

parkerlreed, 2022-02-22:
Great news! (Kinda...)

No longer seems to crash on VTK 9.2 and QT 5.15.2

https://gist.github.com/parkerlreed/09af1075135fd25108a9344607fa0552

I'll close this out for now.

I'll have to open a new ticket as now dragging the visualizer window to move around the scene is dragging the entire window...

closed 2022-02-22 by parkerlreed