introlab/rtabmap #761: Tutorial "export raster layers to meshlab" cannot works

takerfume, 2021-08-09:
I tried to do the exact same things of tutorial ["export raster layers to meshlab"](https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab
), but it does not work.

I used the same data `2loops_workspace_3IT.db ` used in this tutorial and do almost same things of this tutorial.
(I tried to do the exact same thing but the version of meshlab is different between mine and this tutorial. )

I cannot get texture.png after executing "Parametrization+texturing from registered rasters", and also cannot see texture applied on mesh.

Do anyone know how to complete this tutorial with latest version of meshlab and rtabmap?

My environment is here.
```
OS: Ubuntu 18.04
meshlab( result of apt-cache policy meshlab):
  Installed: 1.3.2+dfsg1-4
  Candidate: 1.3.2+dfsg1-4
  Version table:
 *** 1.3.2+dfsg1-4 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status

ros-melodic-rtabmap-ros( result of apt-cache policy ros-melodic-rtabmap-ros):
  Installed: 0.20.9-2bionic.20210505.050942
  Candidate: 0.20.9-2bionic.20210505.050942
  Version table:
 *** 0.20.9-2bionic.20210505.050942 500
        500 http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
```

This is the dialog of meshlab.
```
Opened mesh /home/[username]/.ros/tutorial/bundler/mesh.ply in 213 msec
All files opened in 213 msec
Applied filter Select non Manifold Edges in 31 msec
Applied filter Delete Selected Vertices in 17 msec
VISIBILITY CHECK: 0.453 sec.
BOUNDARY OPTIMIZATION: 0.648 sec.
CLEANING ISOLATED TRIANGLES: 0.056 sec.
* 1764 triangles cleaned.
PATCH EXTRACTION: 0.039 sec.
* 3422 patches extracted, 3383 null patches.
PATCH EXTENSION: 0.087 sec.
PATCHES UV COMPUTATION: 0.056 sec.
PATCH MERGING: 0.073 sec.
* Area reduction: 92.3%.
* Patches number reduced from 3422 to 889.
PATCH TEXTURE PACKING: 2.654 sec.
TEXTURE PAINTING: 0.112 sec.
Applied filter Parameterization + texturing from registered rasters in 4479 msec

```

takerfume, 2021-08-09:
This is my procedure.

## Export to MeshLab

1. Download this database: 2loops_workspace_3IT.db (This is the exactly same as tutorial)
2. Open RTAB-Map ( version  0.20.9)
3. Open the download database 2loops_workspace_3IT.db
I clicked "ok" in this dialog.
[image]
4. Click on "Download all clouds (update cache)". Click OK for "Global map optimized".(This is the exactly same as tutorial)
5. PostProcessing by 'Tools-> Post-processing': This is slightly different from Tutorial.
This is my condition.
[image]

This is tutorials (older version of RTAB Map)
[image]

6. set preference by "Window -> Preferences" like this (the structure of menu is different but the value of parameters are same)
[image]

7. Export bundler like this
[image]
Dialog shows like this.
[image]

8. I expported .out file
9. and I checked there exist cameras.out and list.txt
10. set preference by "Window -> Preferences" like this (the menu is slightly different)
My dialogue
[image]

Tutorial
[image]

11. Click on File->"Export 3D clouds...".  (The latter half is hidden in tutorial)
First half of my procedure.
[image]
Latter half of my procedure.
[image]
12. I save mesh.ply and this is the dialog when finished.
[image]

## Import in MeshLab
1. Open MeshLab ( version 1.3.2)
2. Click "Open project" (This is the exactly same as tutorial)
3. Open `cameras.out`(This is the exactly same as tutorial)
4. Open `list.txt` (This is the exactly same as tutorial)
5. Import `mesh.ply` (This is the exactly same as tutorial)
Output of Show Dialog is here.
```
Opened mesh /home/taketo/.ros/detailed_tutorial/mesh.ply in 235 msec
```
6. Delete non-manifold edge like this (the same as tutorial)
[image]

click apply in the menu when selecting non manifold edges
[image]

7. Generate the texture
[image]

8. click apply
[image]

9. result. Sometimes texture is generated but the texture is totally white, sometimes texture is not generated.

matlabbe, 2021-09-03:
Just followed all instructions of the [tutorial](https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab) and it seems to be still working:
[image]
[image]

This is on Ubuntu 20.04, Meshlab 202003, latest rtabmap version. For new rtabmap versions, we should make sure that all settings are set to default before loading the database. I added this step at the beginning.

What does look like the texture image (in case it is a Meshlab problem)? here is mine:
[image]