introlab/rtabmap #763: Cannot compile with latest g2o

VisionaryMind, 2021-08-10:
RTABMap is looking for looking for older methods in g2o code:

```
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::VertexSBAPointXYZ::VertexSBAPointXYZ()'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::OptimizableGraph::Edge::clone() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*)'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::OptimizableGraph::Vertex::clone() const'
```
These do not appear to be implemented in the current g2o release (20201223_git). Please indicate which version of g2o should be used for compatibility with the latest RTABMap. All of our apps are compiled with march=native, so the Ubuntu Focal packaged g2o will not do.

Also, the latest Focal upgrade has eliminated RTABMap from its default repositories. I am sure you are already aware of this, but wanted to make mention in case there is a backup strategy for obtaining the apt package for the binaries.

VisionaryMind, 2021-08-10:
I have been able to workaround this problem by compiling g2o version 2017, however, RTABMap still generates segmentation faults when attempting to display point clouds. The following libraries have all been compiled with `-march=native`:

- Eigen 3.3.9
- g2o 2017
- PCL 1.11

Are there any other libraries that require march=native to avoid the segmentation fault? Again, it is not possible any longer to install RTABMap with the Ubuntu apt system --- compilation from source remains the only workable option.

parkerlreed, 2021-08-11:
You need g2o master and not the latest release

Currently working for me g2o-git r1305.a0585942-1 (May 19th commit form this year)

So anything newer should be fine.

rlabs-oss, 2021-09-03:
Tried today building rtabmap with g2o (last git log entry 27th July 2021) and cmake failed with lots of g2o related errors.

matlabbe, 2021-09-05:
rtabmap binaries should be available: http://repositories.ros.org/status_page/ros_noetic_default.html?q=rtabmap, also see them here: http://packages.ros.org/ros/ubuntu/dists/focal/main/binary-amd64/Packages

Last month I did some raspberry images and I didn't have problems (because it was arm?). Looks like rtabmap binaries cannot be downloaded on my laptop (Ubuntu 20.04/noetic):
```
Err:1 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-rtabmap amd64 0.20.10-1focal.20210716.092319
  404  Not Found [IP: 64.50.236.52 80]
Err:2 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-rtabmap-ros amd64 0.20.10-1focal.20210716.102632
  404  Not Found [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-rtabmap/ros-noetic-rtabmap_0.20.10-1focal.20210716.092319_amd64.deb  404  Not Found [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-rtabmap-ros/ros-noetic-rtabmap-ros_0.20.10-1focal.20210716.102632_amd64.deb  404  Not Found [IP: 64.50.236.52 80]
E: Unable to fetch some archives, try running apt-get update or apt-get --fix-missing.
```
I tested other packages, and found that ros-noetic-octomap works, but not ros-noetic-octomap-rviz-plugins:
```
Err:1 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-octomap-rviz-plugins amd64 0.2.3-1focal.20210716.094152
  404  Not Found [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-octomap-rviz-plugins/ros-noetic-octomap-rviz-plugins_0.2.3-1focal.20210716.094152_amd64.deb  404  Not Found [IP: 64.50.236.52 80]
E: Unable to fetch some archives, try running apt-get update or apt-get --fix-missing.
```

@VisionaryMind are you aware of this problem with other ros packages?

For g2o, I could make rtabmap compatible with latest g2o official release, but some people use latest version from source, which create other problems. As g2o doesn't have version in the library, it is not possible to make the code compatible with all g2o versions at the same time. I make sure (with some tricks) at least that rtabmap is supporting g2o binaries from Ros melodic and Ros noetic. @rlabs-oss I cannot reproduce your cmake errors with latest g2o version (July 27 2021), rtabmap builds correctly.

matlabbe, 2021-09-05:
It seems that in a docker image, I can fetch `ros-noetic-rtabmap` and `ros-noetic-rtabmap-ros`, there is maybe something wrong with my apt-get on my laptop. 
```
$ docker pull introlab3it/rtabmap:focal
$ docker run -it --rm introlab3it/rtabmap:focal
# apt update
# apt install ros-noetic-rtabmap-ros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  ros-noetic-rtabmap
The following NEW packages will be installed:
  ros-noetic-rtabmap ros-noetic-rtabmap-ros
0 upgraded, 2 newly installed, 0 to remove and 149 not upgraded.
Need to get 8080 kB of archives.
After this operation, 44.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-rtabmap amd64 0.20.10-1focal.20210727.071209 [4157 kB]
Get:2 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-rtabmap-ros amd64 0.20.10-1focal.20210727.081513 [3923 kB]
Fetched 8080 kB in 2s (3482 kB/s)                 
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package ros-noetic-rtabmap.
(Reading database ... 101712 files and directories currently installed.)
Preparing to unpack .../ros-noetic-rtabmap_0.20.10-1focal.20210727.071209_amd64.deb ...
Unpacking ros-noetic-rtabmap (0.20.10-1focal.20210727.071209) ...
Selecting previously unselected package ros-noetic-rtabmap-ros.
Preparing to unpack .../ros-noetic-rtabmap-ros_0.20.10-1focal.20210727.081513_amd64.deb ...
Unpacking ros-noetic-rtabmap-ros (0.20.10-1focal.20210727.081513) ...
Setting up ros-noetic-rtabmap (0.20.10-1focal.20210727.071209) ...
Setting up ros-noetic-rtabmap-ros (0.20.10-1focal.20210727.081513) ...
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...

```

matlabbe, 2021-09-05:
@rlabs-oss while compiling rtabmap with different versions of some libraries, I got those errors:
```
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::OptimizableGraph::Edge::clone() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::OptimizableGraph::Vertex::clone() const'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::VertexSBAPointXYZ::VertexSBAPointXYZ()'
/usr/bin/ld: ../../../bin/librtabmap_core.so.0.20.13: undefined reference to `g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*)'
```
This was because on a previous build rtabmap linked against g2o binaries on ros noetic, then I changed to latest g2o version. Somewhat rtabmap didn't rebuild the following files, the solution is to touch them to make sure cmake rebuild them:
```
find rtabmap/corelib/src/optimizer -type f -exec touch {} +
```

matlabbe, 2021-09-06:
@VisionaryMind For `-march=native`, all libraries depending on EIgen should be rebuilt with that flag. Can you show your cmake info output (from `cmake ..`)? Here are the main ones:

- libpointmatcher
- g2o
- gtsam
- PCL