introlab/rtabmap #764: Mapping with Multiple Camera without sync

jacklu333333, 2021-08-11:
Hi, I am using a **kinetic camera** and a**n Astra camera** for multi-camera mapping with rtabmap. I reference `demo_two_kinects.launch` to set up the launch file. 

However, I experience some IO overhead, when I enable the node `rtabmap_ros/rgbd_sync`. With one sync enable, the publish rate of 2nd camera raw data(before sync) becomes **22hz** (**30hz** is the theoretical rate). If I enable both `rtabmap_ros/rgbd_sync` nodes the 2nd camera raw output rate becomes **0hz**. 

Like the above, I have no choice but to skip the process of sync. 

Hence, I am wondering whether rtabmap support multiple cameras without sync in advance? 

If it is possible is there any sample launch file?

Thanks in advance. 

Best regards,
Jack Lu

parkerlreed, 2021-08-11:
Are you using a Kinect V1 or V2? 

USB overhead for the newer Kinects is very high and if both cameras are on the same USB controller, you could be running out of bandwidth between the two. 

In the case of the Kinect, the recommended fix is dedicating a USB controller per camera. This can be achieved in a few ways.

1. One camera on the motherboard USB 3 and the other on a PCIe card
2. There are single PCIe cards that dedicated a controller to every USB port

When you see the 0Hz condition, do the cameras appear to error out/have any messages related to USB?

(If I'm completely on the wrong track, disregard :) )

jacklu333333, 2021-08-13:
Hi @parkerlreed, 
Thank you for your fast reply I manage to solve this problem by disabling the registered topic for both cameras.

I am using V1 and both my USB ports are USB 3.0. 

What confuses me the most is that when I enable the registered topic the CPU or memory is not overwhelmed but the output is 0 Hz. I guess this occurred due to the bandwidth between CPU and memory is not enough for registered and sync at the same time.

Best Regards,
Jack Lu

closed 2021-08-13 by jacklu333333